CharlesGillanders/homeassistant-alphaESS #97: Time-zone issue with "total load"

ZeeDK, 2023-12-30:
It seems that all the "daily sensors" work as expected (resetting at 00:00) except for "total load", it resets at 01:00.
So I suspect some kind of time-zone issue. I am in GMT+1 (winter time).

An example here from 00:30 is seen below:
[image]

So "grid to load", "discharge" etc. don't have any issues but this one does.

Poshy163, 2023-12-31:
This is known about, unfortunately the devs seem pretty stubborn about total load being able to be calculated by the end user, not them.

JollyG88, 2024-06-03:
I have the same issue. I'm in GMT +10 and my stats don't reset each day till then. Which then screws up all my dashboards

Poshy163, 2024-06-04:
Are you on the newest version of the HACS add-on (4.8)? 

this should of been fixed

JollyG88, 2024-06-05:
Yes, I’m currently on v4.8.

JollyG88, 2024-09-18:
Haven't gotten back to looking at this till recently. Currently just living with it as is. will try and spend a bit more time digging around this weekend as its still annoying me

Poshy163, 2024-09-18:
What version are you on? i dont have this issue anymore. try updating 0.5.5

JollyG88, 2024-09-22:
> What version are you on? i dont have this issue anymore. try updating 0.5.5

Unfortunately updating to 0.5.5 hasn't resolved it for me. Unless there's something else I need to do as well?

Poshy163, 2024-09-23:
Hmm interesting, wonder if your inverter is thinking its on a different timezone? 

total_load comes from the getSumDataForCustomer call in the OpenAPI, which doesnt take in any parameters (eg any date from the integration) 

what inverter you got? and do you know any way to check it is in the same timezone as you are in? not in daylight saving or somehow in UTC+0:30?

JollyG88, 2024-09-23:
So, the data is getting pulled directly from the inverter on my AlphaESS battery. Looking at the web console for my inverter it’s set to my Timezone here in Brisbane and the time restrictions I have set up for when it can discharge all work correctly. 

It just seems to be this integration with HA. I can recreate the integration if I need to to see if that fixes it.

Poshy163, 2024-09-23:
Yeah right, odd. i mean there isnt much we can do i dont think, something must be thinking you are in ACST, rather than AEST? 

could try to recreate the integration, but seems odd? 

coz the integration is calling
`resource = f"{BASEURL}/getSumDataForCustomer?sysSn={sysSn}"`
`return await self.api_get(resource)`

No time/date data is being sent, and the timestamp is being generated from calling time.time(), which would be the current time/date that your NTP (time server) has set it to in HAOS, which should be brissy

Im not trying to blame you here, but it seems to be something on your end? haven't seen anyone else with any issues since it was originally fixes a bit ago? you could try contacting alpahess themselves and see what they say? could be some mismatch caused by the installer

JollyG88, 2024-09-23:
Yeah all good. Completely get it’s bizarre and highest chances are it’s something in my setup. I’m just scratching my head cos of how weird it is. 😂

markx64, 2025-09-24:
I can confirm I have this same problem here in New Zealand. My Total Load data resets to zero at 12:00pm (midday) not 12:00am (midnight). I'm using v0.7.1. All my other Alpha ESS sensors reset at midnight. Does anyone else have this issue? Or found a resolution?

Medicalmusings, 2025-10-01:
> Same issue for me, in Brisbane Australia.
> Screenshot_20251002-075529.png (view on web)

Not new. Been like that for me since the integration started. (Brisbane). On the alpha device itself our times change by an hour when rest of the country goes into daylight saving. Suspect it’s a Timezone bug 🐛 n the alpha system itself. --Allan Palmer Sent from my iPhone

Poshy163, 2025-11-02:
Amazing to see, glad to see it seems it has been fixed since the 1st of nov

closed 2025-11-02 by Poshy163