CharlesGillanders/homeassistant-alphaESS #100: Happy New Year…

Cushion1, 2023-12-31:
Happy New Year all!
I’m guessing that I’m waiting for the world to catch up before I get data again - a once a year bug affecting those down under more?

[image]

Whilst I’m here, a huge thanks to Charles and all the community for continuing to make this work. 
I’m sure this will resolve itself when you all catch up to 2024! 😀
Ian

Poshy163, 2023-12-31:
Happy new year to you as well, 

This is a dupe of [#99](https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/99)

Going to wait and see what happens when midnight UTC hits and go from there

thospe, 2023-12-31:
Happy new year to all AlphaEss-HomeAssistant fans.

dan-s-github, 2024-01-01:
Happy New Year to all of you struggling with AlphaESS

I start hating AlphaESS, thought that was a well engineered system. But then my Generation 2 Smile-5 Inverter broke down and the local importer replaced it with a pre-used Generation 1 Inverter (reseller in liquidation) where I can not easily get access to the modbus connector and a we now have this API issue I can not quickly switch to use a modbus integration.
Okay I must acknowledge that they now at least have a public API and hopefully we get the current issues resolved.

Poshy163, 2024-01-01:
Its a shame, as there is next to no acknowledgement of the issue from the dev working on it. 

i have sent another message in the slack channel, so just gotta wait for him to respond

Lich2023, 2024-01-01:
The whole Integration doesnt Work for me since midnight..initialisation hang.

Poshy163, 2024-01-01:
Yep, stems from this [#57](https://github.com/alphaess-developer/alphacloud_open_api/issues/57)

We could probably improve the logging of it to make it more transparent when the API goes/is down, but there is unfortunately not much that can be done

Lich2023, 2024-01-01:
Any information, when  it will be "fixed"? Or any other possibilities to integrate an Alpha ESS in ha?

CharlesGillanders, 2024-01-01:
All we can do is wait for alpha to fix things - all of their web systems are broken and unfortunately we cannot do anything until they fix them.  Look at the issue 57 noted earlier in this thread to see if there are any updates from alpha.

Cushion1, 2024-01-03:
Seems to have settled down now!

closed 2024-01-03 by Cushion1