CharlesGillanders/homeassistant-alphaESS #98: "Alpha ESS: Set Battery Charge" service not working

braindeadmalc, 2023-12-31:
The service "Alpha ESS: Set Battery Charge" appears to run OK but does not set the Charge settings on the Alpha B3
This was working last time I used it 
All entities can be read fine which implies the API user validation is still OK but then it should be, I've not changed it.

Poshy163, 2023-12-31:
This is due to a change upstream, not due to the API. It is now set to one change every 24 hours.

See here [Interface restriction announcement
](https://github.com/alphaess-developer/alphacloud_open_api/issues/54)

braindeadmalc, 2023-12-31:
> This is due to a change upstream, not due to the API. It is now set to one change every 24 hours.
> 
> See here [Interface restriction announcement ](https://github.com/alphaess-developer/alphacloud_open_api/issues/54)

Thanks for this. It's a seriously good reason not to buy Alpha.
Really not impressed.

Poshy163, 2023-12-31:
I guess they just had to draw a line between faults to convince.

They store that information in flash memory that degrades rapidly when changing data often. And I don't they ever thought that people would be changing it quite so regularly.

braindeadmalc, 2023-12-31:
I guess this Issue could be closed now?

Poshy163, 2023-12-31:
I agree. 

Cc/ @CharlesGillanders

closed 2023-12-31 by CharlesGillanders

haardman, 2024-01-11:
Alpha have stated that changes are now limited to once every 10 seconds and yet I still don't seem to be able to make any changes from HA.  Do I need to do anything