CharlesGillanders/homeassistant-alphaESS #99: unsupported operand type(s) for -: 'NoneType' and 'NoneType'

jamiergrs, 2023-12-31:
Since midnight on January 1 the entire system is unavailable. Checking logs I’m getting the below 

`This error originated from a custom integration`.

```
Logger: custom_components.alphaess
Source: helpers/update_coordinator.py:300
Integration: Alpha ESS (documentation, issues)
First occurred: 12:00:46 AM (300 occurrences)
Last logged: 4:53:34 AM

Unexpected error fetching alphaess data: unsupported operand type(s) for -: 'NoneType' and 'NoneType'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 300, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 56, in _async_update_data
    inverterdata.update({"Solar to Load": _pv - _feedin})
                                          ~~~~^~~~~~~~~
TypeError: unsupported operand type(s) for -: 'NoneType' and 'NoneType'
```

Poshy163, 2023-12-31:
I'm getting the same.... I have a feeling (if you not on the beta) it has something to do with the alpha API, as I'm also having some trouble with the app (and logging in). 

Have mentioned it in the slack channel, just waiting for a response 

If this persists in a few hours I'll see what workaround may be.

Poshy163, 2023-12-31:
Getting the same, will wait a bit and see what comes from it.

They may have changed a few endpoints, but will have to investigate

jamiergrs, 2023-12-31:
This error is also appearing, but by the looks of it has been appearing the last few days 

```
This error originated from a custom integration.

Logger: alphaess.alphaess
Source: custom_components/alphaess/coordinator.py:30
Integration: Alpha ESS (documentation, issues)
First occurred: December 31, 2023 at 9:28:18 AM (1267 occurrences)
Last logged: 6:49:22 AM

Unexpected json_response : {'code': 6026, 'msg': 'System error', 'expMsg': "Query failed with error code 202 with name 'NetworkInterfaceExceededTimeLimit' and error message 'Encountered non-retryable error during query :: caused by :: Couldn't get a connection within the time limit' on server 10.0.0.22:22001; nested exception is com.mongodb.MongoQueryException: Query failed with error code 202 with name 'NetworkInterfaceExceededTimeLimit' and error message 'Encountered non-retryable error during query :: caused by :: Couldn't get a connection within the time limit' on server 10.0.0.22:22001", 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getOneDateEnergyBySn?sysSn=(removed)&queryDate=2024-01-01
Unexpected None: returned when calling https://openapi.alphaess.com/api/getOneDateEnergyBySn?sysSn=(removed)&queryDate=2024-01-01
Unexpected json_response : {'code': 6026, 'msg': 'System error', 'expMsg': "Query failed with error code 202 with name 'NetworkInterfaceExceededTimeLimit' and error message 'Encountered non-retryable error during query :: caused by :: Couldn't get a connection within the time limit' on server 10.0.0.32:22001; nested exception is com.mongodb.MongoQueryException: Query failed with error code 202 with name 'NetworkInterfaceExceededTimeLimit' and error message 'Encountered non-retryable error during query :: caused by :: Couldn't get a connection within the time limit' on server 10.0.0.32:22001", 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getSumDataForCustomer?sysSn=(removed)
Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=(removed)
Unexpected json_response : {'code': 6026, 'msg': 'System error', 'expMsg': "Query failed with error code 202 with name 'NetworkInterfaceExceededTimeLimit' and error message 'Encountered non-retryable error during query :: caused by :: Couldn't get a connection within the time limit' on server 10.0.0.22:22001; nested exception is com.mongodb.MongoQueryException: Query failed with error code 202 with name 'NetworkInterfaceExceededTimeLimit' and error message 'Encountered non-retryable error during query :: caused by :: Couldn't get a connection within the time limit' on server 10.0.0.22:22001", 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getSumDataForCustomer?sysSn=(removed)
```

Poshy163, 2024-01-01:
Ah shit. The intergration won't initialise at all. 

Will look into it when I get into my laptop

jamiergrs, 2024-01-01:
It will sporadically re-initialise but only load the instantaneous entities but eventually the whole integration will fail and then not initialise again

Poshy163, 2024-01-01:
Got any logs?

jamiergrs, 2024-01-01:
These logs are from the two errors that seem to be generating from just now: 

```
`This error originated from a custom integration.

Logger: custom_components.alphaess
Source: helpers/update_coordinator.py:300
Integration: Alpha ESS (documentation, issues)
First occurred: 4:53:15 AM (6 occurrences)
Last logged: 8:20:03 AM

Unexpected error fetching alphaess data: 'NoneType' object is not iterable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 300, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 30, in _async_update_data
    jsondata: json = await self.api.getdata()
                     ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/alphaess/alphaess.py", line 301, in getdata
    for unit in units:
TypeError: 'NoneType' object is not iterable
```

And this one 

```
This error originated from a custom integration.

Logger: alphaess.alphaess
Source: custom_components/alphaess/coordinator.py:30
Integration: Alpha ESS (documentation, issues)
First occurred: December 31, 2023 at 9:28:18 AM (1330 occurrences)
Last logged: 8:20:03 AM

Error: when calling https://openapi.alphaess.com/api/getChargeConfigInfo?sysSn=AL7002223050974
Cannot connect to host openapi.alphaess.com:443 ssl:default [Connection reset by peer]
Error: Cannot connect to host openapi.alphaess.com:443 ssl:default [Connection reset by peer] when calling https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=AL7002223050974
Error: Cannot connect to host openapi.alphaess.com:443 ssl:default [Connection reset by peer] when calling https://openapi.alphaess.com/api/getEssList
'NoneType' object is not iterable
```

jamiergrs, 2024-01-01:
And now these 

```

Logger: homeassistant.config_entries
Source: config_entries.py:402
First occurred: 7:25:56 AM (3 occurrences)
Last logged: 8:28:51 AM

Error setting up entry alpha9e44edeb0ceaaf48 for alphaess
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 402, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/__init__.py", line 48, in async_setup_entry
    await coordinator.async_config_entry_first_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 265, in async_config_entry_first_refresh
    await self._async_refresh(
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 300, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 30, in _async_update_data
    jsondata: json = await self.api.getdata()
                     ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/alphaess/alphaess.py", line 304, in getdata
    unit ['SumData'] = await self.getSumDataForCustomer(serial)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/alphaess/alphaess.py", line 103, in getSumDataForCustomer
    data = await self.__get_data(resource)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/alphaess/alphaess.py", line 236, in __get_data
    response = await session.get(
               ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/client.py", line 601, in _request
    await resp.start(conn)
  File "/usr/local/lib/python3.11/site-packages/aiohttp/client_reqrep.py", line 965, in start
    message, payload = await protocol.read()  # type: ignore[union-attr]
                       ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/streams.py", line 622, in read
    await self._waiter
asyncio.exceptions.CancelledError

```

Poshy163, 2024-01-01:
I can't tell if it's an issue on our end or theirs? 

I'm on my way home now, so will investigate it a bit more

ettiennecoetzee, 2024-01-01:
I've tried to reinstall, but I can't even complete the setup now?

Poshy163, 2024-01-01:
Right, using my postman collection i've got, nothing is working. (just returning 500 server error and/or a timeout)

Ive sent a message in the dev slack channel, so just gotta wait for a response. don't remove the integration if you already have it linked, it will just intermittently try to reconnect and restart it.  

Will update this again when i get some more meaningful update

Poshy163, 2024-01-01:
This is the response, not much we can do....

[image]

ZeeDK, 2024-01-01:
My Alpha ESS app is working again, it was failing earlier today. Had hoped the integration would then also be up and running again, but it seems that it is still not getting any data :(

dan-s-github, 2024-01-01:
My AlphaESS app is currently working but I can not login to the alpha cloud website so there seems to still be some issues.
Also my AlphaESS App was also working when I checked yesterday and I was able to disable the charge from grid, as it was set to charge before the API changes broke the integration.

Poshy163, 2024-01-01:
Yep, no doubt the app using a internal API to gather data, not the same one we have access to.

Ill work on a better implementation of checking if the API isnt responding, so the whole integration doesn't hang in initializing. but in the meantime this whole ordeal is rather disappointing, and the lack of communication for what makes it even worse

ettiennecoetzee, 2024-01-01:
Have you guys tried local control with this? https://github.com/dxoverdy/Alpha2MQTT

CharlesGillanders, 2024-01-02:
> Have you guys tried local control with this? https://github.com/dxoverdy/Alpha2MQTT 

Not personally but I've had messages from someone who has used this and is very happy with it.  I just didn't want to have to bother with the hardware setup if the web approach was reliable.

jamiergrs, 2024-01-02:
I was just able to add the integration back into HA and all the sensors have come back online. Potentially Alpha have now resolved their issue?

ettiennecoetzee, 2024-01-02:
I am also back online.

dan-s-github, 2024-01-02:
> Have you guys tried local control with this? https://github.com/dxoverdy/Alpha2MQTT

I was planning to try implementing a version of this using [ESPHOME and an ATOM RS-485 KIT](https://github.com/dxoverdy/Alpha2MQTT/issues/11#issuecomment-1742362560) and contacted the author. 
Unfortunately I currently [don't have access](https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/100#issuecomment-1873240013) to the modbus interface and can't do anything .

can confirm that my system was able to retrieve data again at 18:21:17 UTC+13
```
2024-01-02 18:21:17.025 INFO (MainThread) [custom_components.alphaess] Fetching alphaess data recovered
```

no change regarding the getDisChargeConfigInfo 
```
2024-01-02 18:22:19.837 ERROR (MainThread) [alphaess.alphaess] Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=XXXXXXXXXXXX
2024-01-02 18:23:22.165 ERROR (MainThread) [alphaess.alphaess] Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=XXXXXXXXXXXX
2024-01-02 18:24:25.243 ERROR (MainThread) [alphaess.alphaess] Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=XXXXXXXXXXXX
```

matt-oneill, 2024-01-02:
I'm still getting errors connecting to the open api in postman and home assistant, seems they haven't fixed everything yet.
Have tried resetting developer key as that worked before but still not right

```
Logger: alphaess.alphaess
Source: custom_components/alphaess/config_flow.py:28
Integration: Alpha ESS ([documentation](https://github.com/CharlesGillanders/homeassistant-alphaESS), [issues](https://github.com/CharlesGillanders/homeassistant-alphaESS/issues))
First occurred: 10:47:46 (2 occurrences)
Last logged: 10:47:46

Unexpected json_response : {'code': 6009, 'msg': 'The request timestamp is not within the validity period'} when calling https://openapi.alphaess.com/api/getEssList
'NoneType' object is not iterable
```

djsepp, 2024-01-04:
Alpha ess announced yesterday that the limit of 1x every 24 hours will be raised. However, you have expressed yourself somewhat confusingly as to how often you can now set the value for charging / discharging. 
https://github.com/alphaess-developer/alphacloud_open_api/issues/54

philosifer, 2024-01-04:
I'm getting a 404 to https://openapi.alphaess.com/ web page with a browser currently and the api shows the same. 

`[Thu, 04 Jan 2024 14:05:42] ERROR [alphaess.py.__post_data:290] 404, message='', url=URL('https://openapi.alphaess.com/api/updateChargeConfigInfo')`

The Android app and the alpha cloud website are both fine though. Seems like the whole API is down again.

closed 2024-01-11 by CharlesGillanders

CrazyHarry69, 2024-06-20:
> {'code': 6009, 'msg': 'The request timestamp is not within the validity period'} when calling https://openapi.alphaess.com/api/getEssList

did you get any results fixing this

CharlesGillanders, 2024-06-20:
> > {'code': 6009, 'msg': 'The request timestamp is not within the validity period'} when calling https://openapi.alphaess.com/api/getEssList
> 
> 
> 
> did you get any results fixing this

That's not something I can likely fix, it's probably either a problem with Alpha ESS API service or your home assistant device is not synchronised properly to an accurate time source.  

Do you get the same error trying to access Alpha ESS via postman or similar?

CrazyHarry69, 2024-06-25:
hi team. still having problems with trying to get the Alpha ESS HAC working.  THe 6009 error resolved itself but now im getting the below. Im only a basic level so struggling through this. any help would be greatly appreciated 

Logger: alphaess.alphaess
Source: custom_components/alphaess/config_flow.py:28
integration: Alpha ESS ([documentation](https://github.com/CharlesGillanders/homeassistant-alphaESS), [issues](https://github.com/CharlesGillanders/homeassistant-alphaESS/issues))
First occurred: 13:29:20 (2 occurrences)
Last logged: 13:29:20

    Unexpected json_response : {'code': 6026, 'msg': 'network is busy', 'expMsg': 'For input string: "H0"', 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getEssList
    'NoneType' object is not iterable