CharlesGillanders/homeassistant-alphaESS #101: 404 Error

DavidqStokes, 2024-01-04:
Yesterday evening and this evening I've received a 404 error when my automation runs to set my AlphaESS battery overnight charge times.

Is anyone else seeing this??

(Last night I was still on HA 2013.12.x and today I'm on 2024.01.1)

This error originated from a custom integration.

Logger: alphaess.alphaess
Source: custom_components/alphaess/__init__.py:57
Integration: Alpha ESS (documentation, issues)
First occurred: 22:45:01 (2 occurrences)
Last logged: 22:45:01

404, message='', url=URL('https://openapi.alphaess.com/api/updateChargeConfigInfo')
Error: 404, message='', url=URL('https://openapi.alphaess.com/api/updateChargeConfigInfo') when calling https://openapi.alphaess.com/api/updateChargeConfigInfo

DaveBinM, 2024-01-05:
Yes, I’m seeing the same thing now.

Poshy163, 2024-01-05:
Upstream issue here [#59](https://github.com/alphaess-developer/alphacloud_open_api/issues/59)

closed 2024-01-11 by CharlesGillanders