CharlesGillanders/homeassistant-alphaESS #103: Deprecation warning in HA 2024.1.2

dan-s-github, 2024-01-07:
Happy New Year and some new deprecations ;-)

After updating HA core to 2024.1.2 I noticed following deprecation warnings

```
2024-01-07 12:53:14.656 WARNING (MainThread) [homeassistant.const] ENERGY_KILO_WATT_HOUR was used from alphaess, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfEnergy.KILO_WATT_HOUR instead, please create a bug report at https://github.com/CharlesGillanders/homeassistant-alphaESS/issues
2024-01-07 12:53:14.670 WARNING (MainThread) [homeassistant.const] POWER_WATT was used from alphaess, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfPower.WATT instead, please create a bug report at https://github.com/CharlesGillanders/homeassistant-alphaESS/issues
```

Poshy163, 2024-01-07:
Yeah, its been fixed with the latest pull: [Updated deprecated measurement usage](https://github.com/CharlesGillanders/homeassistant-alphaESS/pull/102)

CharlesGillanders, 2024-01-11:
fixed in v0.4.8

closed 2024-01-11 by CharlesGillanders