CharlesGillanders/homeassistant-alphaESS #104: Doesn't work from midnight until 1p.m.

ahmedhoumeci, 2024-01-15:
Hi,
I have the following error and I don't know how to solve it:

**Logger: custom_components.alphaess
Sourc**e: helpers/update_coordinator.py:300
Integration: Alpha ESS (documentation, issues)
First occurred: 14 gennaio 2024 alle ore 01:09:45 (956 occurrences)
Last logged: 07:55:09

Unexpected error fetching alphaess data: unsupported operand type(s) for -: 'NoneType' and 'NoneType'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 300, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 56, in _async_update_data
    inverterdata.update({"Solar to Load": _pv - _feedin})
                                          ~~~~^~~~~~~~~
TypeError: unsupported operand type(s) for -: 'NoneType' and 'NoneType'**

And the integration doesn't work from midnight until 1 p.m. more or less.

Any suggestion?
Thanks

Poshy163, 2024-01-15:
what does your API response look like during that time? i recently had the same issue, but was caused due to one of the sites being offline (cant connect to the internet), therefore everything returned as "none" in the API response

See here:
[image]

Im working on a fix to catch this an alert the end user, could you be having the same issue?

ahmedhoumeci, 2024-01-15:
@Poshy163 thank you for your answer.
I can reach the cloud site without any problems if is that your question.

Poshy163, 2024-01-15:
Apologies, let me rephrase.

During 12-1 am;
- Does the integration start up at all? or does it fail to boot?
- Is your AlphaESS device connected to the internet during that hour? 
- What does the logs show during that hour? do you get a returned API call in the logs? (something like this?)
[image]

If you are unsure, you can just send the logs through and i can have a look at what might be causing the issue

ahmedhoumeci, 2024-01-16:
Hi @Poshy163. To answer to your question:
1. The integration failed to boot
2. Yes, it is correctly connette to internet
3. I don't find anything in the logs exept this:

2024-01-15 09:34:00.056 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data: unsupported operand type(s) for -: 'NoneType' and 'NoneType'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 300, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 56, in _async_update_data
    inverterdata.update({"Solar to Load

ahmedhoumeci, 2024-01-16:
@Poshy163 the entire error logs of my home assistant istance
[error_log-4.txt](https://github.com/CharlesGillanders/homeassistant-alphaESS/files/13955913/error_log-4.txt)

Poshy163, 2024-01-17:
@ahmedhoumeci during that time, was your system connected to the internet? 
As i stated recently i had the same issue, but was caused due to one of the sites being offline (cant connect to the internet), and that threw me the "supported operand type(s) for -: 'NoneType' and 'NoneType'" error 

if you know how to program/edit the source file, do you want to play a debug output for "feedin" and "_pv", as i have a feeling they are "none". if not dont stress as i can show you how a bit later.

i am currently away, but i arrive in antwerp tomorrow so i can work on a bit more of the debugging/bypass because it seems like a common issue that some people have

ahmedhoumeci, 2024-04-20:
Hi @Poshy163 did you find a workaround for the issue?

closed 2024-05-04 by ahmedhoumeci

Poshy163, 2024-07-26:
@ahmedhoumeci can you try the new beta release and see if it has been resolved? seems to be fixed on my end (from midnight to next API call) 

(to get the beta, go to the HACS page for AlphaESS, click on the 3 dots at the top right. click show beta versions, and download the latest beta. restart)