CharlesGillanders/homeassistant-alphaESS #109: NoneType Log Error

beastien, 2024-03-04:
I keep getting this error in my logs just after midnight.
What would be causing this?

This error originated from a custom integration.

Logger: custom_components.alphaess
Source: helpers/update_coordinator.py:313
Integration: Alpha ESS (documentation, issues)
First occurred: 12:00:38 AM (4 occurrences)
Last logged: 12:03:45 AM

Unexpected error fetching alphaess data: unsupported operand type(s) for -: 'NoneType' and 'NoneType'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 313, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 56, in _async_update_data
    inverterdata.update({"Solar to Load": _pv - _feedin})
                                          ~~~~^~~~~~~~~
TypeError: unsupported operand type(s) for -: 'NoneType' and 'NoneType'

Poshy163, 2024-03-05:
What timezone are you in? and do you have any data missing/gaps for a certain period of time after midnight?

it has something to do from trying to the data for the current date, but alpha cant produce that information (due to their wack systems) and it throws that issue. 

this should of been resolved a while ago. but maybe there is some edge cases.

vk1kcm, 2024-03-08:
I'm having the same problem.  Timezone is UTC+11 (Australian AEST Daylight Savings).  It was working fine until 4th March, then there are gaps, now it just throws the error.

Poshy163, 2024-03-08:
Hmm strange, I'm based in UTC +10:30 (central daylight) and not having an issue. I'll do some debugging tommrow to work out if it's an alpha issue, or something with the API somehow.

vk1kcm, 2024-03-09:
Getting this as well.
```
This error originated from a custom integration.

Logger: alphaess.alphaess
Source: custom_components/alphaess/coordinator.py:30
integration: Alpha ESS (documentation, issues)
First occurred: March 8, 2024 at 7:52:55 PM (20 occurrences)
Last logged: 1:32:22 PM

Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getChargeConfigInfo?sysSn=AL8001321010696
Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=AL8001321010696
Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getChargeConfigInfo?sysSn=AL8001321010580
Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=AL8001321010580
```

vk1kcm, 2024-03-12:
Ok, I have two systems.  One of them was not reporting.  I got it reconnected to the wifi and reporting about an hour ago, and the plugin started working again.

beastien, 2024-03-12:
I'm also in Timezone UTC+11 (Australian AEST Daylight Savings). I'm seeing a gap in the data from 12:00-12:05am

Poshy163, 2024-03-13:
Yeah, i had the same issues with my 2 systems, when one goes down. the whole integration fails, i tried to decouple the two systems but unfortunately with how the API pulls the data it isnt the easiest without rewriting the whole intergration. maybe one day ill have the motivation to do it. 

As for the 6053 error, it apparently relates to the request being too fast. which no doubt is just the API being called too many times, as for each system you have registered via HA, it makes that call at almost the same time, which could limit it. 
ill add some code to suppress to/let you know about it not log as an error, as it doesn't seem to have cause any data gaps. 

As for the gap in data from 12-12:05, i don't seem to have that issue (seem to have a unbroken line through that time), you got any logs from around then and that might help to get a better idea of what is going on around then