CharlesGillanders/homeassistant-alphaESS #111: TypeError: unsupported operand type

ahmedhoumeci, 2024-05-04:
How can I solve it?
I saw differents people with the same issue, but I didn't understand the solution.
Here is the log:
Questo errore ha avuto origine da un'integrazione personalizzata.

Registratore: custom_components.alphaess
Fonte: helpers/update_coordinator.py:315
Integrazione: Alpha ESS (documentazione, problemi)
Prima occorrenza: 00:00:22 (428 occorrenze)
Ultima registrazione: 07:37:43

Unexpected error fetching alphaess data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 56, in _async_update_data
    inverterdata.update({"Solar to Load": _pv - _feedin})
                                          ~~~~^~~~~~~~~
TypeError: unsupported operand type(s) for -: 'NoneType' and 'NoneType'

ahmedhoumeci, 2024-05-04:
On https://openapi.alphaess.com/ I'm getting this: {"code":6012,"msg":"AppId is empty"}

ahmedhoumeci, 2024-05-26:
Guys, could you help me please?

CharlesGillanders, 2024-05-26:
Are you able to download the alphaessOpenAPI python library for yourself?  

https://github.com/CharlesGillanders/alphaess-openAPI

There's a test python script in the library which you could use to try to see where the API call is going wrong.

ahmedhoumeci, 2024-05-26:
> Are you able to download the alphaessOpenAPI python library for yourself?
> 
> https://github.com/CharlesGillanders/alphaess-openAPI
> 
> There's a test python script in the library which you could use to try to see where the API call is going wrong.

Unfortunately, I'm not able 🫠

ahmedhoumeci, 2024-05-26:
It's getting frustrating to have the integration not working from 12am to 2pm

Poshy163, 2024-05-27:
>  not working from 12am to 2pm

Are you able to manually reset the intergration? does that work? 

>  {"code":6012,"msg":"AppId is empty"}

this just means that the no appID is being added to the requests, this error was common when the integration was moved from Web scraping to the OpenAPI, can you delete the integration and re-add it?

ahmedhoumeci, 2024-05-27:
> Are you able to manually reset the intergration? does that work?
>
>
> this just means that the no appID is being added to the requests, this error was common when the integration was moved from Web [...]

I already do it and doesn't work.

I will delete the integration and re-add it

ahmedhoumeci, 2024-05-27:
@Poshy163 it doesn't work, also if I delete it and re-add it.
Any other suggestion?

ahmedhoumeci, 2024-05-27:
[image]

This is what I got. I also delete and re-add the system on open api...but still not work.

Poshy163, 2024-05-29:
yeah right, 

While this seems to be an issue with the alpha API and its jank with the return values, the debugging and logging could no doubt be better, im working on a fixing most of the issues with the code, (as a single issue like this shouldn't halt the entire integration, rather log and make the entities unavailable). might take a little while tho as its basically a big jigsaw puzzle

ahmedhoumeci, 2024-05-29:
Thank you so much @Poshy163 
Surely your "workaround" will help me and I'll not be anymore struggled 😅

ahmedhoumeci, 2024-05-29:
The strange thing that I don't understand is why after a certain hour, the integration return to work properly

Poshy163, 2024-06-04:
Made a fork here (https://github.com/Poshy163/homeassistant-alphaESS) and will merge it into the main branch here once its resolved, seems like it basically need a whole re-write of it to have better error handling 

i have included the openAPI as another file (so you are not importing it) allowing anyone to edit it in case it *might* be helpful to change a few logic stuff in the API

ahmedhoumeci, 2024-06-04:
So, what I need to do to make it work again?

Poshy163, 2024-06-04:
Nothing at the moment 😅, im testing it locally (with more debugging and better logs) and will make a PR to this repo once i reckon its fixed/better.

CharlesGillanders, 2024-06-28:
> Nothing at the moment 😅, im testing it locally (with more debugging and better logs) and will make a PR to this repo once i reckon its fixed/better.

I just took a quick look at your fork, Home Assistant as a rule requires integrations to import their library functions from PyPi rather than including them in their code.  If I ever wanted to submit the integration for inclusion as one of the built in integrations then importing the openapi library code is required.  I'm really interested in any suggestions you have for better code but the library should be kept separate from the integration, at least once ready for the pull request anyway....

Poshy163, 2024-06-28:
yeah nah i know that, just testing it so i can also made modifications independent of the actual API if needed (which i doubt). testing it locally hence why the repo hasn't been updated in a while, learning how home assistant integrations run from scratch and how you've implemented it takes a while 🤣 

But from what ive gathered, how the integration handles the information it receives when the API returns it as "blank" isnt the best, and it throws this error (but doesn't actually break the data logging in HA i think?) and spits this error

ahmedhoumeci, 2024-07-02:
Guys, any news? I would love to be able to use it.
I don't understand if it's just me that has this problem 😢

CharlesGillanders, 2024-07-02:
> Guys, any news? I would love to be able to use it.
> 
> I don't understand if it's just me that has this problem 😢

Unfortunately the problem only happens with some systems. My own system has no problems so it's really hard for me to debug - I need to see what is coming back from Alpha for your system which I cannot do unless you are able to download the python library and run the test script yourself.

Adding extra debug code is important and would potentially help in finding the problem but the fastest way to solve this would be for you to be able to send the test script output.

Poshy163, 2024-07-02:
double this, i also don't have a issue besides from like 5 minutes after midnight (which has been covered is as a result of them sending back blank data and im working on better improving) 

No doubt this is some issue around the response the API is giving you, but without the raw response, its hard to know how to tackle it 😅

ahmedhoumeci, 2024-07-04:
Thanks guys for the answers.
If you hace time and patience, you can explain me how to download the python library and how to run the test script so I can senza the test script 😊😊😊

CharlesGillanders, 2024-07-15:
> Thanks guys for the answers.
> 
> If you have time and patience, you can explain me how to download the python library and how to run the test script so I can share the test script 😊😊😊

Apologies for the slow reply.  Actually I think there's an easier way than walking you through installing the library and running it at the command line.

A couple of excellent contributions from other developers have produced a Postman collection which is available here https://github.com/CharlesGillanders/alphaess-openAPI/blob/dee30e03e685be394f31399140a9857dfffd5c86/AlphaESS%20Open%20API.postman_collection.json

Please download the raw json file and save it to your local PC you can then connect to postman.com and import the collection.

I'll post a follow up message here on how to use it to query Alpha's API and get results back.

CharlesGillanders, 2024-07-15:
> 
> I'll post a follow up message here on how to use it to query Alpha's API and get results back.

Once you have the collection imported into postman you should click on the name of the collection where it says "AlphaESS OpenAPI" - that should open a window in the main view where you can see various tabs, you need to edit the variables tab.

In the variables tab you need to edit the Initial Value and Current Value fields for AppID, AppSecret, and SysSN.

The SysSN is the serial number of your invertor and the AppID and AppSecret are your AppID and AppSecret from [https://open.alphaess.com/](https://open.alphaess.com/), once you have set all three variables in both current and initial value hit the save button.

Then you should run each of the API calls in turn - one after the other and paste the results here, removing your SysSN for confidentiality.

for example - this is what my invertor returns for GetEssList, if you can post all of you results then we should be able to start seeing where your system is different.

{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": [
        {
            "sysSn": "[removed]",
            "popv": 12.30,
            "minv": "SMILE5-INV",
            "poinv": 5.00,
            "cobat": 11.40,
            "mbat": "SMILE5-BAT",
            "surplusCobat": 10.9,
            "usCapacity": 90.00,
            "emsStatus": "Normal"
        }
    ],
    "extra": null
}

CharlesGillanders, 2024-07-15:
Oh PS - I don't need you to run the POST API calls in postman, only the GET calls are any value in debugging this issue.

ahmedhoumeci, 2024-07-16:
Thank you very much @CharlesGillanders , I will try and get back to you

Poshy163, 2024-07-20:
@ahmedhoumeci how did you go? any luck getting the results of it?

homeconnexde, 2024-07-21:
getesslist
{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": [
        {
            "sysSn": "--",
            "popv": 9.60,
            "minv": "Storion-S5",
            "poinv": 4.60,
            "cobat": 9.00,
            "mbat": "M4860",
            "surplusCobat": 8.1,
            "usCapacity": 90.00,
            "emsStatus": "Normal"
        }
    ],
    "extra": null
}
getsumdata
{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": {
        "epvtoday": 46.16,
        "epvtotal": 17003.14,
        "eload": 8.40,
        "eoutput": 37.01,
        "einput": 0.55,
        "echarge": 3.40,
        "edischarge": 2.10,
        "todayIncome": 12.92,
        "totalIncome": 4263.64,
        "eselfConsumption": 0.47,
        "eselfSufficiency": 0.52016,
        "treeNum": 2911.1488,
        "carbonNum": 14622.7004,
        "moneyType": "EUR"
    },
    "extra": null
}
getlastpowerdata
{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": {
        "ppv": null,
        "ppvDetail": null,
        "soc": null,
        "pev": null,
        "pevDetail": null,
        "prealL1": null,
        "prealL2": null,
        "prealL3": null,
        "pgrid": null,
        "pgridDetail": null,
        "pbat": null,
        "pload": null
    },
    "extra": null
}
getonedaypowerbysn
{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": [
        {
            "sysSn": "---",
            "uploadTime": "2024-07-21 20:09:17",
            "ppv": 312.00,
            "load": 466.0,
            "cbat": 100.0,
            "feedIn": 0,
            "gridCharge": 101.0,
            "pchargingPile": 0
        },
        {
            "sysSn": "---",
            "uploadTime": "2024-07-21 20:04:19",
            "ppv": 321.00,
            "load": 437.0,
            "cbat": 100.0,
            "feedIn": 0,
            "gridCharge": 64.0,
            "pchargingPile": 0
        }, ... repeated data for every 5 minutes approximately

getonedateenergybysn
{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": {
        "sysSn": "---",
        "theDate": "2024-07-21",
        "eCharge": 3.40,
        "epv": 46.16,
        "eOutput": 37.01,
        "eInput": 0.55,
        "eGridCharge": 0.00,
        "eDischarge": 2.10,
        "eChargingPile": 0.00
    },
    "extra": null
}

getchargeconfiginfo
{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": {
        "gridCharge": 0,
        "timeChaf1": "00:00",
        "timeChae1": "13:00",
        "timeChaf2": "00:00",
        "timeChae2": "00:00",
        "batHighCap": 100.00
    },
    "extra": null
}
getdischargeconfiginfo
{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": {
        "ctrDis": 0,
        "timeDisf1": "00:00",
        "timeDise1": "00:00",
        "timeDisf2": "00:00",
        "timeDise2": "00:00",
        "batUseCap": 10.00
    },
    "extra": null
}

closed 2024-07-24 by CharlesGillanders

Poshy163, 2024-07-24:
@homeconnexde now it has been merged and added to the beta branch (here) https://github.com/CharlesGillanders/homeassistant-alphaESS/releases/tag/v0.5.0beta

are you able to go to the HACS page for AlphaESS, click on the 3 dots at the top right. click show beta versions, and download the latest beta. restart, and see if that works?

PR #116 (2024-07-24): Storion-S5 System fix and Improvements
TODO: 

- Add additional data points provided by the API.
- Check if any data points from the API are blank or a whitespace, if so. default to 0 (should fix #111 and #112)
- If multiple inverters exists, throttle the subsequent inverters calls to reduce the chance of a 6053 max requests error 
....