CharlesGillanders/homeassistant-alphaESS #112: Integration not initialising

homeconnexde, 2024-05-21:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 82, in _async_update_data
    inverterdata.update({"Instantaneous PPV1": _pvpowerdetails.get("ppv1")})
                                               ^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'

dhmusik, 2024-06-26:
Same Prob

Poshy163, 2024-07-21:
@homeconnexde @dhmusik are you able to follow the steps in #111 (down the bottom from Charles and post what result you get?

homeconnexde, 2024-07-21:
Just posted the postman results in https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/111

CharlesGillanders, 2024-07-21:
> Just posted the postman results in https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/111
> 
> 

Fascinating - no real time power values returned at all.  I presume in the website and/or mobile app you are getting real time power?

CharlesGillanders, 2024-07-21:
I had a look in the Alpha ESS github, there's an old issue there about storion S5 systems not providing the same real time power data that other Alpha units do.

https://github.com/alphaess-developer/alphacloud_open_api/issues/49

I'll have to rewrite the home assistant integration to not expect real time power for some systems.

homeconnexde, 2024-07-21:
Yes in the App their ist a power value, but the update interval is 5 Minutes.

homeconnexde, 2024-07-21:
Thank you very much for your effort.

Poshy163, 2024-07-21:
Shouldn't be too hard, maybe its best to just call the coordinator first; pre-sensor start up and then use one of 2 "sets" of sensors depending on if its Storion-S5 based or not. ill see if i can work something out

Poshy163, 2024-07-24:
@homeconnexde an attempt at a fix has been made in #116, just waiting for the pull into the main branch

CharlesGillanders, 2024-07-24:
> @homeconnexde an attempt at a fix has been made in #116, just waiting for the pull into the main branch

@homeconnexde, @Poshy163 

v0.5.0beta was just published, HACS should have it available if you've elected to try beta releases.

If you haven't previously elected to try beta releases see this from HACS - https://hacs.xyz/docs/faq/beta/

homeconnexde, 2024-08-19:
Ok, now version 0.5.2 is installed.

I can add an entry with AppID and Secret but it still says. initialisation not possible. no devices or entitys found for the AppID

Poshy163, 2024-08-20:
Yeah, has been fixed here: #122, just pending the pull 

Also, you mind showing me the model name? should be something in the logger that has includes your serial number, and model `_LOGGER.info(f"Serial:{serial}, Model: {model}")`
(where {serial} is the serial number of the inverter, you can blank this out, and {model} is the model, which i need)

you should be able to find it by opening the logs, Settings > System > Logs > Load all logs. then doing control + F and searching for model. should be a bit like this
[image]

homeconnexde, 2024-08-20:
I don't have such information in the log. It's just the following two things:
AttributeError: 'NoneType' object has no attribute 'get'

2024-08-20 07:03:22.063 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 90, in _async_update_data
    inverterdata["Instantaneous PPV1"] = await process_value(_pvpowerdetails.get("ppv1"))
                                                             ^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'

and sometimes:

AttributeError: 'NoneType' object has no attribute 'get'
2024-08-19 21:33:18.347 ERROR (MainThread) [alphaess.alphaess] Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getChargeConfigInfo?sysSn=AK1510120700059
2024-08-19 21:33:18.516 ERROR (MainThread) [alphaess.alphaess] Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=AK1510************
2024-08-19 21:33:18.516 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
self.data = await self._async_update_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/alphaess/coordinator.py", line 90, in _async_update_data
inverterdata["Instantaneous PPV1"] = await process_value(_pvpowerdetails.get("ppv1"))
^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'

homeconnexde, 2024-08-20:
Ok the changed coordinator.py from #122 works. The initialisation was successful. I got now 1 device and 14 entities. 

Thank you very much.

But the entity State_of_charge is missing.

Poshy163, 2024-08-20:
Wonderful! glad to see that patch is working 😅, ill get charles to merge it asap 

as for state of charge, do you mind using the postman collection (found here: https://github.com/CharlesGillanders/alphaess-openAPI/blob/main/AlphaESS%20Open%20API.postman_collection.json) and posting the results for the `getLastPowerData` GET request? it might be something different, tho. currently both `State of Charge` and `Instantaneous Battery SOC` are linked to the "battery %"
[image]

Poshy163, 2024-08-20:
Along that, if you see any values in any of those requests you would like to see added to the integration (or you just want to see what might be added), you can make a new issue with a dump of all of the GET requests and i can have a look if there is anything different from what we have. (neither me or the main maintainer have the older inverters/batteries)