CharlesGillanders/homeassistant-alphaESS #113: Detected blocking call to open inside the event loop by custom integration 'alphaess' at custom_components/alphaess/coordinator.py, line 30

PjotrBaas, 2024-06-08:
Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/setup.py", line 165, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 447, in _async_setup_component await asyncio.gather( File "/usr/src/homeassistant/homeassistant/setup.py", line 449, in <genexpr> create_eager_task( File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked await self.async_setup(hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/config/custom_components/alphaess/__init__.py", line 48, in async_setup_entry await coordinator.async_config_entry_first_refresh() File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 277, in async_config_entry_first_refresh await self._async_refresh( File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 312, in _async_refresh self.data = await self._async_update_data() File "/config/custom_components/alphaess/coordinator.py", line 30, in _async_update_data jsondata: json = await self.api.getdata()

Spitfiredriver, 2024-06-15:
Same warning after recent update.
Logger: homeassistant.util.loop
Source: util/loop.py:84
First occurred: 09:36:24 (24 occurrences)
Last logged: 09:37:34

Detected blocking call to open inside the event loop by custom integration 'alphaess' at custom_components/alphaess/coordinator.py, line 30: jsondata: json = await self.api.getdata() (offender: /usr/local/lib/python3.12/netrc.py, line 74: with open(file, encoding="utf-8") as fp:), please create a bug report at https://github.com/CharlesGillanders/homeassistant-alphaESS/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 239, in __wrap_handle_refresh_interval self.config_entry.async_create_background_task( File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1164, in async_create_background_task task = hass.async_create_background_task(target, name, eager_start) File "/usr/src/homeassistant/homeassistant/core.py", line 860, in async_create_background_task task = create_eager_task(target, name=name, loop=self.loop) File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 255, in _handle_refresh_interval await self._async_refresh(log_failures=True, scheduled=True) File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 312, in _async_refresh self.data = await self._async_update_data() File "/config/custom_components/alphaess/coordinator.py", line 30, in _async_update_data jsondata: json = await self.api.getdata()
Detected blocking call to open inside the event loop by custom integration 'alphaess' at custom_components/alphaess/coordinator.py, line 30: jsondata: json = await self.api.getdata() (offender: /usr/local/lib/python3.12/netrc.py, line 74: with open(file, encoding="utf-8") as fp:), please create a bug report at https://github.com/CharlesGillanders/homeassistant-alphaESS/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 255, in _handle_refresh_interval await self._async_refresh(log_failures=True, scheduled=True) File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 312, in _async_refresh self.data = await self._async_update_data() File "/config/custom_components/alphaess/coordinator.py", line 30, in _async_update_data jsondata: json = await self.api.getdata()

wallus, 2024-06-21:
I am getting this as well: 

2024-06-21 14:38:08.555 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open inside the event loop by custom integration 'alphaess' at custom_components/alphaess/coordinator.py, line 30: jsondata: json = await self.api.getdata() (offender: /usr/local/lib/python3.12/netrc.py, line 74: with open(file, encoding="utf-8") as fp:), please create a bug report at https://github.com/CharlesGillanders/homeassistant-alphaESS/issues
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 255, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 312, in _async_refresh
    self.data = await self._async_update_data()
  File "/config/custom_components/alphaess/coordinator.py", line 30, in _async_update_data
    jsondata: json = await self.api.getdata()

CharlesGillanders, 2024-06-21:
This is a warning from Home Assistant, it doesn't like something that I am doing when retrieving the data from AlphaESS, it thinks it may be blocking the event loop for other integrations.  I am not clear on why the warning is coming up, all of the functions I call are being called asynchronously so they should not be blocking anything else.

The couple of other forum posts that I saw with similar issues seem to be related to actually synchronous API calls and if I try to use the recommended solution for those the integration breaks entirely.  I've asked on the development forum to see if anyone has any suggestions, in the meantime all I can suggest is that you ignore those warnings until I have a solution.

wallus, 2024-06-25:
Thanks for the update. I can confirm that there does not seem to be any negative impact on my system except for the log message spamming.

CharlesGillanders, 2024-06-28:
I just pushed out update 0.4.9 which fixes the issue that was causing the blocking warning.

closed 2024-06-28 by CharlesGillanders