CharlesGillanders/homeassistant-alphaESS #115: Documentation of values in ha

Mjimsas, 2024-07-03:
Hey, this is just a general question but is there any documentation/information regarding the meaning of the values that get exposed in HA?

CadeusTheGreat, 2024-07-07:
Are you able to elaborate? I could put together a docs page for what all the values mean if that would be helpful. i.e. total load, state of charge, charge, discharge etc... If that's the values you're referring to. :)

Mjimsas, 2024-07-08:
Yes exactly that. For my use case i found the values i needed. I found that the values exposed in HA (naming) are different from the ones written in the api docs from alphaESS themselves.

Mjimsas, 2024-07-08:
For my use case i wanted to have the current discharge in W. But didn't know which value to use for this. In the end i think that this sensor value exposed by the integration: 
[image]

Seems to correspond with this one in the alphaESS api docs.
[image]

For me it was guess work and a bit of logical thinking to figure out what value was the one i actually needed.

CharlesGillanders, 2024-07-08:
I'm working on the assumption that Instantaneous Load is the total demand of the house on the inverter.  

That demand could be partly met by the grid, partly met by solar generation and partly met by battery discharge.  

The pload value is indeed what I'm using but we may have different interpretations of what the alpha documentation means....

closed 2024-08-02 by CharlesGillanders