CharlesGillanders/homeassistant-alphaESS #121: Entity sensor.goanna_solar_to_battery from integration alphaess has state class total_increasing, but its state is negative.

crayner, 2024-08-12:
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:326
integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 11:30:10 (1 occurrences)
Last logged: 11:30:10

Entity sensor.goanna_solar_to_battery from integration alphaess has state class total_increasing, but its state is negative.

Perhaps the state class should be measurement (or total), not total_increasing.

Poshy163, 2024-08-12:
How is your solar to battery negative? seems to be a rounding issue or something on alpha's end, that number should always be positive? 

What inverter have you got, and how often does it have a negative value?

crayner, 2024-08-12:
This was as per the initial installation of the AlphaESS Integration.  The integration is at fault if it does not handle a rounding error.  The figure has been stuck at -0.2 kWh since the integration installation.  The invertor is a Storion-SMILE-B3.  It never showes a negative value on the Alpha ESS Monitoring page.

Poshy163, 2024-08-12:
we just get the data from the API, and show them, not the duty of the integration to show "correct" information, rather just the information fed through the openAPI.

All i can suggest is delete that warning. and then just if it comes up again, check to see in the the logs if that value is shown (aka its being fed that through the API calls), and then re-open this issue or make a new one

crayner, 2024-08-12:
I reset the value to 0 in the UI.  Reporting in log stopped.

closed 2024-08-12 by CharlesGillanders