CharlesGillanders/homeassistant-alphaESS #124: SOC and other values missing on Storion-S5

homeconnexde, 2024-08-20:
Hi Charles,

the SOC, load, PPV and other data from the Storion-S5 is **not in the getLastPowerData**:

{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": {
        "ppv": null,
        "ppvDetail": null,
        "soc": null,
        "pev": null,
        "pevDetail": null,
        "prealL1": null,
        "prealL2": null,
        "prealL3": null,
        "pgrid": null,
        "pgridDetail": null,
        "pbat": null,
        "pload": null
    },
    "extra": null
}

it is only in the **getOneDayPowerBySn** 

{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": [
        {
            "sysSn": "AK1510**********",
            "uploadTime": "2024-08-20 09:59:20",
            "ppv": 5642.00,
            "load": 666.0,
            "cbat": 33.2,
            "feedIn": 2690.0,
            "gridCharge": 0,
            "pchargingPile": 0
        },
        {
            "sysSn": "AK1510**********",
            "uploadTime": "2024-08-20 09:54:19",
            "ppv": 5486.00,
            "load": 711.0,
            "cbat": 31.2,
            "feedIn": 2487.0,
            "gridCharge": 0,
            "pchargingPile": 0
        },
        {
            "sysSn": "AK1510**********",
            "uploadTime": "2024-08-20 09:49:22",
            "ppv": 5213.00,
            "load": 677.0,
            "cbat": 28.8,
            "feedIn": 2249.0,
            "gridCharge": 0,
            "pchargingPile": 0
        },

Poshy163, 2024-08-20:
what is `cbat`? that your current battery charge?

homeconnexde, 2024-08-20:
Yes cbat is the SOC

Sorry I forgot to mention that.

Poshy163, 2024-08-20:
Cool, ill add that in. ill link the pull request here when ive done it, i can try to keep sensor name the same, so what would you rather it be called? `State of Charge` or `Instantaneous Battery SOC`? no point having both in there

Poshy163, 2024-08-20:
Ah rough, the root API doesn't call getOneDayPowerBySn at all, i can try a work around, but we are going to need to patch the base API as well. might take a bit to get it up and running 👍🏻

Poshy163, 2024-08-20:
Cool. a draft has been made, will need the API to be updated before you can test it locally. but the pull is at #126