CharlesGillanders/homeassistant-alphaESS #128: Failed to Initialise

strickersuk, 2024-08-31:
Seems to have stopped working this morning, cloud and app working fine.

The only log error I can see is:

TypeError: unsupported operand type(s) for -: 'str' and 'float'

strickersuk, 2024-08-31:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 354, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 84, in _async_update_data
    inverterdata["Solar to Load"] = _pv - _feedin
                                    ~~~~^~~~~~~~~

suds32, 2024-08-31:
Same for me

strickersuk, 2024-08-31:
I'll close this issue to prevent duplication.

closed 2024-08-31 by strickersuk