CharlesGillanders/homeassistant-alphaESS #129: Integration failing to setup "unsupported operand type(s) for -: 'str' and 'float'"

jamiergrs, 2024-08-31:
The integration suddenly failed to start today and no matter if you reset, delete and re-add I'm still getting the following errors; The integration shows the error as "Failed setup, will retry: unsupported operand type(s) for -: 'str' and 'float'"

Checking the logs the following can be seen
`Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getChargeConfigInfo?sysSn=[removed]`

`Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=[removed]`

And addtionally 
`Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 354, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 84, in _async_update_data
    inverterdata["Solar to Load"] = _pv - _feedin
                                    ~~~~^~~~~~~~~
TypeError: unsupported operand type(s) for -: 'str' and 'float'`

Not sure what's changed but haven't had any issues with the integration or month, app still works fine

strickersuk, 2024-08-31:
Same for me, I opened issue 128 just before you.

Rapt0rx, 2024-08-31:
The same issue started this morning on my system.

bmannz, 2024-08-31:
Yes same issue for me also. Started 1hr ago for me.

suds32, 2024-08-31:
Same for me as well

strickersuk, 2024-08-31:
I close #128 to prevent duplication.

strickersuk, 2024-08-31:
I commented out line 84 in coordinator.py and have most of it back. Hopefully a quick fix for someone much smarter than me.

crayner, 2024-08-31:
Seeing the same here.

Also in logs: Logger: alphaess.alphaess
Source: custom_components/alphaess/coordinator.py:41
integration: Alpha ESS ([documentation](https://github.com/CharlesGillanders/homeassistant-alphaESS), [issues](https://github.com/CharlesGillanders/homeassistant-alphaESS/issues))
First occurred: 15:22:20 (2 occurrences)
Last logged: 15:22:20

Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getChargeConfigInfo?sysSn=xxxxxxxxxxxxxxxx
Unexpected json_response : {'code': 6053, 'msg': 'The maximum number of requests has been reached', 'expMsg': None, 'data': None, 'extra': None} when calling https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=xxxxxxxxxxxxxxx

Logger: custom_components.alphaess
Source: helpers/update_coordinator.py:354
integration: Alpha ESS ([documentation](https://github.com/CharlesGillanders/homeassistant-alphaESS), [issues](https://github.com/CharlesGillanders/homeassistant-alphaESS/issues))
First occurred: 15:21:11 (34 occurrences)
Last logged: 15:57:41

Unexpected error fetching alphaess data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 354, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 70, in _async_update_data
    inverterdata["Solar to Load"] = await process_value(_pv - _feedin)
                                                        ~~~~^~~~~~~~~
TypeError: unsupported operand type(s) for -: 'str' and 'float'

Hope this helps!  Craig

jamiergrs, 2024-08-31:
> I commented out line 84 in coordinator.py and have most of it back. Hopefully a quick fix for someone much smarter than me.

This also worked for me, can get all data back except obviously the Grid to Load value

Poshy163, 2024-08-31:
What da fuckkkkkkk, I'll look into it now

TheSimTek, 2024-08-31:
Same here. Was fine until o restarted HA

Poshy163, 2024-08-31:
Ok, an update for yall, seems like for some fucking reason epv is a string?!?!?! No idea why this happened can only imagine it was a mistake?

[image]

a hot fix solution is to set line 78 of coordinator.py from 

`_pv = await safe_get(_onedateenergy, "epv")` 

to 

`_pv = float(await safe_get(_onedateenergy, "epv"))`

this basically forces the string to be a float

techmaved, 2024-08-31:
It seems like the api has been fixed. The integration works as usual.

Poshy163, 2024-08-31:
Yeah, appears to be fixed now, might need a integration refresh to pull the latest API changes 

[image]

DaveBinM, 2024-08-31:
Although it’s come back, my total_load sensor is now reporting as 0, which isn't right. Curious if anyone else sees this?

jamiergrs, 2024-08-31:
> Although it’s come back, my total_load sensor is now reporting as 0, which isn't right. Curious if anyone else sees this?

Mine is also showing as 0 since the integration started failing yesterday

Poshy163, 2024-09-01:
Huh yeah just noticed that. eload, which we use for the total load, is showing as 0 from the API 😢 
[image]

Most of this is our of our control unfortunately, clearly they have messed with the API last night and it has had some side effects...

Poshy163, 2024-09-01:
it has been mentioned upstream: https://github.com/alphaess-developer/alphacloud_open_api/issues/82

closed 2024-09-06 by CharlesGillanders