CharlesGillanders/homeassistant-alphaESS #132: Wrong values in the energy dashboard from AlphaESS system

Meigl89, 2024-09-03:
Hi everyone, 

hopefully you can help me with this mess, or maybe somebody encountered the same problem. 

Starting from 1st September I'm facing the issue that the energy dashboard shows a complete mess regarding my solar values. 

[image]

The reality is this:

[image]

I didn't change anything here for weeks, why should I though, because it was running super smooth. 

Yesterday it was also wrong:

[image]

versus reality:

[image]

And for today it will be the same, since its wrong even in the morning hours it shows 4,5kwh of load, in reality its 2,4kwh. 

I honestly dont get it. I thought it came from the battery, because values here are super crazy, but when I remove the battery from the dashboard, the other values are still wrong. 

Funny - or sadly - the entities are completely correct. For yesterday prodcution was 33,3kwh und entitiy says:

[image]

I'm really struggling in finding the root cause of this.. 

Can anybody give me a hint what may be wrong here?

Thank you all!

Poshy163, 2024-09-03:
Heya, the issue has been mentioned here: https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/127 and also upstream here https://github.com/alphaess-developer/alphacloud_open_api/issues/82

TL:DR Its a combination of them giving invalid data (through their server errors and long time outs on requests), and us defaulting to using the value of 0 when receiving invalid data. which is causing these spikes, 

A draft PR is open to fix this here: https://github.com/CharlesGillanders/homeassistant-alphaESS/pull/130

Meigl89, 2024-09-03:
Oh I didn't put that together, that this may be the same problem. Thank you for the information regarding this! 

So I can do nothing but wait, thats fine, now I know that it's not on my end. Hopefully it can be fixed. 

Thank you very much!

Euan-p, 2024-09-03:
Same problem here. anyone know if the root API problem is being worked on by alphaess or  have we just got the custom PR? Wasn't sure if someone might have found a forum post about the api that i missed.

ElevenFan, 2024-09-05:
I didn't modify anything in my program，I suggest you find after-sales service to give feedback.

closed 2024-09-06 by CharlesGillanders