CharlesGillanders/homeassistant-alphaESS #134: Failing to initialize

tbgoose, 2024-09-16:
Hi all

I was having similar issues to others but thought I would create a new issue as I have already undertaken debug steps and have some info to include to help others. The integration stopped functioning about 3-4 days ago. Tried reloading etc but it is stuck on initializing. I have enabled debug logging but not seeing much honestly in the logs. Only one error RE connectivity at midnight.

[image]

[image]

My machine can curl openapi url
```
➜  ~ curl https://openapi.alphaess.com/
{"code":6012,"msg":"AppId is empty"}#     
```
However using postman I receive the following result on all get calls. Google tells me this is likely an authentication issue, but possible could relate to the timestamp format once everything is encrypted together in the api calls? [Link ](https://github.com/alphaess-developer/alphacloud_open_api/issues/12). Anyhow, I have reset the developer secret to try get around that, but still receive the same results in postman.
[image]

I have now also uninstalled the integration completely, restarted HA, reinstalled 0.5.5, restarted HA and tried to re-add integration using the new appsecret and **it is finally back up and working**. Not sure why Postman isn't working, perhaps I am doing something incorrectly.

Hopefully this helps someone else.

ceeessbee, 2024-09-18:
Hi all, also having problems when reloading HA, I was looking to move to new version but have now reverted back as no entities available. Not as technically literate as @tbgoose but have followed his approach to delete AlphaESS, restart, re-login. However I now can't find AlphaESS through HACS (all kinds of searching) so have created manually. I've reset my AlphaESS 'secret' and setup using original ID and new 'secret'. However I get a screen response of "Unknown error occurred". I'm lost on where to go with this now. As @tbgoose implies maybe an AlphaESS authintication issue at the server end which may clear (wishful thinking).

Poshy163, 2024-09-18:
Yeah, their server seems to be down. its going to be a issue for everyone unfortunately. 

Even through using postman to make the API calls, you get an error 😢 

[image]

ceeessbee, 2024-09-18:
Thank you for confirming @Poshy163

beastien, 2024-09-18:
I'm having the same problem.

This error originated from a custom integration.

Logger: alphaess.alphaess
Source: custom_components/alphaess/__init__.py:48
integration: Alpha ESS (documentation, issues)
First occurred: September 18, 2024 at 10:51:21 PM (4 occurrences)
Last logged: 5:31:02 AM

502, message='Bad Gateway', url='https://openapi.alphaess.com/api/getEssList'
Error: 502, message='Bad Gateway', url='https://openapi.alphaess.com/api/getEssList' when calling https://openapi.alphaess.com/api/getEssList

jetscrubber, 2024-09-18:
Same issue here. ESS cloud is not down, my phone app work properly.

Freddo333, 2024-09-18:
> Same issue here. ESS cloud is not down, my phone app work properly.

It seems that its just the open api thats down. The phone app has been using an internal api for a while now, which annoyingly alpha ess has increased the security on and has a habit of changing, so it hasn't been as practical to use for a while.

Meigl89, 2024-09-19:
Working again on my end. API is back.

[image]

itsgumbys, 2024-09-19:
hi,
i have reinstalled 0.5.5 version using HACS. integration is added, however no entities are being shown.
what could be the reason ?

Regards,

Poshy163, 2024-09-19:
What error are you getting via the logs? have you tried reloading the integration?

itsgumbys, 2024-09-19:
> What error are you getting via the logs? have you tried reloading the integration?

TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'model_id'
2024-09-19 10:57:31.028 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity None for domain sensor with platform alphaess
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 600, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 772, in _async_add_entity
    device = dev_reg.async_get(self.hass).async_get_or_create(

tried reloading no luck !

Poshy163, 2024-09-19:
Huh weird, does this error still occur if you remove the integration and delete it from HACS, restart. then install it again, restart and then try to add the OpenAPI information again?

pjhum, 2024-09-19:
Now the API is back up, a restart of HA had everything running again for me- I had held off doing  anything In the meantime, hoping that would be the case.

Poshy163, 2024-09-19:
@itsgumbys working for me. how many inverters do you have in your openAPI account? and do they show there? 

[image]

itsgumbys, 2024-09-19:
> @itsgumbys working for me, just re-added my 2 inverters. how many inverters do you have in your openAPI account? and do they show there?
> 
> ![image](https://private-user-images.githubusercontent.com/53972874/368889505-0cbf2724-4dde-445c-bf61-77bce733af99.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY3MzE5NTcsIm5iZiI6MTcyNjczMTY1NywicGF0aCI6Ii81Mzk3Mjg3NC8zNjg4ODk1MDUtMGNiZjI3MjQtNGRkZS00NDVjLWJmNjEtNzdiY2U3MzNhZjk5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE5VDA3NDA1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI0ZGZhNWJlNzMzYzkwNmMxMjRmZTU0MDYxMGY0YWMyYzVjM2IzMDQ3NDBjOTYyNGNlYjY4YmU4MGM4Njg3NWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5HNquDpRhk50hmIwSTiCAZBajQGzJB9sfGpInAp1i28)

only 1 inverter and its being shown on openAPI.
all i remember is that i reset the key before adding.

which version of alphaess integration you are using ? may be updating to latest version can caused some issue.

Poshy163, 2024-09-19:
do you have IP whitelist turned on? where you cannot get any data fropm the API on ur IP address? 
and are you using the most current developer id and Developer key? it might take a bit to update with the new key if you have just reset it 

im also on 0.5.5.

itsgumbys, 2024-09-19:
> do you have IP whitelist turned on? where you cannot get any data fropm the API on ur IP address? and are you using the most current developer id and Developer key? it might take a bit to update with the new key if you have just reset it
> 
> im also on 0.5.5.

Well... i have restored HA to last backup and re-entered the key as i have reset it and its back to life now !!. Thanks for the help :) but i dont know why it was behaving.

tbgoose, 2024-09-23:
Mine is now back up. I had done a few integration reloads and some ha restarts for other updates over last few days. Not sure what finally brang it back but restart this morning _fixed it_...