CharlesGillanders/homeassistant-alphaESS #136: Problems retrieving entities / initialising

DavidqStokes, 2024-10-13:
I suspect that this is an AlphaESS issue, however...

Yesterday morning I noticed that some entities (e.g. SOC from my Smile B3 battery and PV production figure from my monitored PV system) were missing. They appeared intermittently during the day and would then disappear.

This morning I noticed that the integration will not initialise (error log below). I also cannot log in to the AlphaESS app on my phone and am thinking that AlphaESS are 'tinkering' this weekend.

Anyone else have problems?

Error log below:
============

Logger: homeassistant.config_entries
Source: config_entries.py:594
First occurred: 08:00:03 (2 occurrences)
Last logged: 10:13:10

Error setting up entry alphabeb5fe159238b3ae for alphaess
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/__init__.py", line 49, in async_setup_entry
    for unit in ESSList:
TypeError: 'NoneType' object is not iterable

blackshroud, 2024-10-13:
I suspect this has to do with the global maintenance being done by AlphaESS (for the last 38 hours the app itself has been intermittently down/not displaying data, however, it might be that the API service itself has been changed/updated.

Lets wait and see once the outage is marked as resolved.

DavidqStokes, 2024-10-13:
Thanks for confirming it's not just me.

I agree that we need to wait and see what happens and respond accordingly. I have significant faith in this development community.

jamiergrs, 2024-10-13:
Was about to ask the same thing, I've had basically all instantaneous data not available and the app showing the device as "offline" but a solid COM light. Visiting cloud.alphaess I can see the device is online but also missing all live data and SOC 

I'm assuming everyone is having similar issues and is an issue with the AlphaESS api

Poshy163, 2024-10-13:
Spot on, due to the current maintenance that has been going on for the last few days. its producing most of the response from the API as "null"
[image]

DaveBinM, 2024-10-13:
Based on this comment, it seems as if it’s an issue with AlphaESS’ servers, rather than an API change. 

https://github.com/alphaess-developer/alphacloud_open_api/issues/88#issuecomment-2408756845

TheSimTek, 2024-10-14:
Same issue here.  App and web back online but cannot initialise the integration.

Poshy163, 2024-10-14:
API is still down, throwing a bunch of errors on the API responses, would be a bunch of "Unexpected error fetching alphaess data" errors in your logs.

They seem to be prioritising getting the app interface fixed first, then moving onto the OpenAPI, whenever that might be

dan-s-github, 2024-10-14:
Interesting strategy to close [Issue 88](https://github.com/alphaess-developer/alphacloud_open_api/issues/88) and then open a new issue [Issue 89](https://github.com/alphaess-developer/alphacloud_open_api/issues/89)

DavidqStokes, 2024-10-14:
This is clearly an AlphaESS issue and I will therefore close this issue when/if the API comes back working okay with the integration.
If it comes back broken we can use this issue to identify what needs to be revised.
Thanks for the input and feedback everyone.

MonsterEnergy-wtf, 2024-10-14:
It starts to piss me off to see that internet relying devices get inaccesible while maintenance. The manufacturer xould lock us oht when ever they want to.

I think I will start to use Modbus to grab data directly from the device with IOBroker rather than using the workaround with the unreliable API...

Not sure why the device cannot be used locally only via LAN... 🤷‍♂️

dan-s-github, 2024-10-14:
Unrelated but I also wanted to swap to modes integration, now my supplier replaced my smile 5v2 with a smile 5v1 where the modbus connector is not easily accessible and I forgot to connect a cable at the time of replacement :-(

puja108, 2024-10-15:
My AlphaESS App is already working again, but the HA plugin is still erroring out. Might still related to the maintenance window as there could different servers/endpoints being used. Just wanted to put it here, in case people were also wondering.

eviltabouleh, 2024-10-16:
I think that the API has changed , I understand that they are still possibly having issues with the platform but I am seeing this error  [alphaess.alphaess] Error:  when calling https://openapi.alphaess.com/api/getEssList
[homeassistant.config_entries] Error setting up entry alphaXXXXXXXXXX for alphaess

I am in Australia and using a VPP (Amber Electric) and they use this API to control our batteries , this started working correctly around 4pm yesterday afternoon

Poshy163, 2024-10-16:
Yeah, im with amber too, weirdly enough thats working. odd, ill reach out to elevenfan and see if anything has changed with the API. super odd. maybe they are web scraping or just not using the openAPI at all?

Meigl89, 2024-10-16:
Now we have the nginx error page. Thats the same we had last time, when there were problems with the API. Fingers crossed.

Poshy163, 2024-10-16:
Appears to be back giving data, but some of it is going to be null.... at least the integration wont error out anymore 
[image]

jamiergrs, 2024-10-16:
My integration still is failing to initialise with the below 
[image]

strickersuk, 2024-10-16:
[image]

DavidqStokes, 2024-10-16:
I tried to reinitialise. All entities except one were Unavailable. The error log shows the following:

Logger: homeassistant.config_entries
Source: config_entries.py:594
First occurred: 08:40:46 (1 occurrences)
Last logged: 08:40:46

Error setting up entry alphabeb5fe159238b3ae for alphaess
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/__init__.py", line 49, in async_setup_entry
    for unit in ESSList:
TypeError: 'NoneType' object is not iterable

blackshroud, 2024-10-16:
I have a new home assistant setup, so trying to setup the add-on for the first time, and I am just getting "an unknown error occured" through the UI when entering app secret etc etc

dan-s-github, 2024-10-16:
The APP no longer shows the maintenance warning, so hopefully everything will resolve soon

lweberru, 2024-10-16:
The integration is now able to intialize again, but most sensors are not filled.

jamiergrs, 2024-10-16:
Couldn't see anything in my logs about cbat. Looks like all data has come back except the "live" data I.e all instantaneous data such as SOC and Load is missing

Poshy163, 2024-10-16:
Nvm. API is still down, all good still spilling null data for getLastPowerData

[image]

dmutsaers, 2024-10-16:
I just checked and it looks like the integration is available again:

[image]

cpmueller, 2024-10-16:
As far as I can see the service is flapping. It is not stable an not all data is available.
We have to wait :-(

dmutsaers, 2024-10-16:
@cpmueller You're right the service is unavailable again.

Poshy163, 2024-10-16:
Im currently working on a new pull [here](https://github.com/CharlesGillanders/homeassistant-alphaESS/pull/135) for some new charge/discharge features. 

I can add some edge cases to the API so that the whole thing doesn't reset/error out when the API doesn't return data, rather just throws an error to the user, but still continues to try and grab new data every 5 mins. 

is that something you'd be interested in seeing? resolves manually having to restart the integration when either your internet goes down and it cant reach the API, or the API refuses to connect (like now)

Meigl89, 2024-10-16:
That would be awesome!

jamiergrs, 2024-10-16:
I'm also just getting null from postman. I'm assuming this will just resolve itself once the data starts coming back through? Seems all requests are passing ok now 
[image]

jamiergrs, 2024-10-17:
Has there been any update? Seems they've removed the issues from the AlphaESS Open API GitHub, can't even see them any more

Poshy163, 2024-10-17:
Yeah, they closed the issues, and are just referring people to contact support about it.... 

[image]

MonsterEnergy-wtf, 2024-10-17:
Awesome.
You pay a shit amoubt of money for the whole system (in my case 25k €) and all you get is a shitty support. Awesome.
I guess I will go for Modbus integration within IOBroker and send data via MQTT to HA. The my whole system will be unplugged from the net.
I'm tired of intransparent support like this. I meN: when the official API is down, how will the owner of this project bring back the integration?

But I guess this is the common way of a lot of service providers. 

Time to change something.

blackshroud, 2024-10-17:
interesting, I cannot even open the open.alphaess.com website anymore....

Just get a 500 error

[image]

eviltabouleh, 2024-10-17:
> interesting, I cannot even open the open.alphaess.com website anymore....
> 
> Just get a 500 error
> 
> ![Screenshot_20241017-181325.png](https://private-user-images.githubusercontent.com/2756699/377362329-1947d714-4628-4305-a6b1-10039ce5e6a9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkxNTMyNDcsIm5iZiI6MTcyOTE1Mjk0NywicGF0aCI6Ii8yNzU2Njk5LzM3NzM2MjMyOS0xOTQ3ZDcxNC00NjI4LTQzMDUtYTZiMS0xMDAzOWNlNWU2YTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAxNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMTdUMDgxNTQ3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTk5OTljZmRkZThmYjU4M2ZjOTZkZjBhMGJlNjkzYjIyN2U5NDFmMTM4MDlkYWQwZmM2N2EwOTYxYzUwMWY5OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.vmtYxOueUMFc84wVPtGSqO8sjmBXcGp7JbidclxRsdM)

It's still flaky and I've noticed that it may need a cache clear for some browsers

strickersuk, 2024-10-17:
I can login to the API website. Weird how some data is showing from the API and some isn't. App and Website seem OK now so I assume it's just a waiting game.

tonbut, 2024-10-17:
The API seems to be back online for me now with getLastPowerData returning correct values.

jamiergrs, 2024-10-17:
Looks like it is returning intermittently. getLastPowerData seems to be pulling values but drops in and out 
[image]

strickersuk, 2024-10-17:
All back now. Quick question slightly off topic, I had my inverter replaced and I now have 2 services. One populated one not, I've removed the entities but would appreciate some help removing the service?

Meigl89, 2024-10-17:
On my end still nothing. But let's wait and see. Last time it also came in batches if I remember correctly..

puja108, 2024-10-17:
It's generally back for me, but I still see some flapping every now and then, which does resolve itself without reload/restart though.

DavidqStokes, 2024-10-17:
I can now initialise the integration, but many values are either reporting 'unknown' or 'unavailable'.
It's almost six days now that this has been unreliable.

speedyhb, 2024-10-17:
Everything is working again !

dry, 2024-10-17:
> Working again for me too. Still also thinking about a Modbus integration
> within IOBroker that sends data via MQTT to HA. Anybody know a good guide
> for the MQTT to HA part?

This works very well https://github.com/dxoverdy/Alpha2MQTT.

DavidqStokes, 2024-10-17:
I now have only one entity reporting unavailable and I'm not 100% sure it's relevant to my battery/inverter.

If it's all good in the morning I will close this issue.

gregpetersau, 2024-10-17:
Conforming finally working for me in Syd Aus too. Took quote a few restarts to get them all, but all entities seem to now report correctly

kaffeetrinker71, 2024-10-18:
the API apparently provides data according to POSTMAN (see screenshot).
But the integration (0.5.5) still does not receive any data.
[image]

DavidqStokes, 2024-10-18:
As this seems to be back to normal I am now going to close this issue, with thanks to everyone who has provided updates throughout.

closed 2024-10-18 by DavidqStokes

Poshy163, 2024-10-18:
@kaffeetrinker71 make a new issue with logs attached if your still having issues after restarting home assistant