CharlesGillanders/homeassistant-alphaESS #138: "Alpha ESS: Set Battery Charge" not setting

braindeadmalc, 2024-10-19:
Automations using "Alpha ESS: Set Battery Charge" don't set the battery charge period on my Smile B3. 
No errors, just not setting the times

These automations worked prior to the latest Alpha upgrade.

Looking at the Alpha API site, they appear to be stating  that setting one per day is allowed which is what caused problems for a lot of us. It got changed to once every 15 mins but I'm wondering if they've put it back to once per day?

mapo-89, 2024-10-19:
I have the same problem, as well as with the discharge

dan22h, 2024-10-19:
Same problem on my Smile5 but it doesn't even work once per day so I don't imagine that's the issue. I really need to convert to a direct MODBUS connection but I can't get it to work on my system for some unknown reason.

mapo-89, 2024-10-19:
But it doesn't work with Postman either. So the problem seems to be with the API

Euan-p, 2024-10-19:
My SMILE5 isn't working either, but only the charge setting for some reason? Is throwing up an error of "unknown action".

DavidqStokes, 2024-10-19:
Just done some testing via my scheduled automation.
Although HA doesn't appear to be throwing any errors into the logs, the update to the SMILE B3 settings fails (as confirmed via the AlphaESS app)

Poshy163, 2024-10-20:
Cheers for the reports guys, I'm working on a new PR to hopefully improve the charging/discharging stuff. Great to have alpha throw some edge cases these last few weeks so I can verify if the whole system works 😅

DavidqStokes, 2024-10-20:
> Cheers for the reports guys, I'm working on a new PR to hopefully improve the charging/discharging stuff. Great to have alpha throw some edge cases these last few weeks so I can verify if the whole system works 😅

And the on-going work on this (by everyone involved) is appreciated as always.

braindeadmalc, 2024-10-20:
I've raised a ticket with Alpha assuming it's their issue.

braindeadmalc, 2024-10-23:
Here's the reply from Alpha

Description Customer message ince no rs/h The Onen APL "Set Charging Information" called from the t long Com prob ome alphi 2424 Com Cu View Andrew Bailey 2024-10-23 08:14:46 
Dear Customer, Thank you for reaching out to us and bringing this to our attention. We are working on this issue and expected to take few days to resolve. If you still experience the issue on 28/10, please create a new support request again. Sorry for any inconvenience it may caused.
 Kind Regards, Eddie Yam Alpha ESS UK Evaluate

braindeadmalc, 2024-10-23:
Here's the reply from Alpha 

We are working on this issue and expected to take few days to resolve. If you still experience the issue on 28/10, please create a new support request again. Sorry for any inconvenience it may caused. Kind Regards, Eddie Yam Alpha ESS UK

DavidqStokes, 2024-10-31:
It's now 31st October and my battery charge / discharge setting still didn't work last night. Are others still having issues?

dan22h, 2024-10-31:
I got fed up with the constant issues caused by the AlphaESS updates and have finally moved to MODBUS which is well worth the effort, the Inverter responds immediately to commands now and unlocks features that aren't possible otherwise, e.g. stopping the batteries charging from solar. 
Thanks for the hardwork carried out making this integration, it was a great help before Alpha ESS messed it up.

Poshy163, 2024-10-31:
> Are others still having issues?

Personally not, what inverter you got? could be related some what to that? the PR i got going [here](https://github.com/CharlesGillanders/homeassistant-alphaESS/pull/135) should give more transparency regarding the currently set charge and discharge times (when the API polls) 

Should give you a little bit of info to give to alpha if its not fixed on your end

braindeadmalc, 2024-10-31:
I still have the problem and have raised the case again as they requested

Poshy163, 2024-11-01:
Hey there, do you have the full name (the model of it) so I can add it to the blacklist? Should show in your logs as Model: xxx Serial: xxxx

tonbut, 2024-11-01:
Thank you everyone for updates, I'm not using this repo but read it to get the good information! Instead I'm using the [alphaess-openAPI](https://github.com/CharlesGillanders/alphaess-openAPI). I have a Smile5 and am having the same issues - reading works, but updates `updateChargeConfigInfo` isn't having any effect despite returning no errors. I hope alpha get their stuff sorted and at a minimum are more transparent about whats going on.

pjhum, 2024-11-01:
Mine is Alpha Smile B3 serial AL7002022040171

Poshy163, 2024-11-02:
Sorry @pjhum should of been more specific. when the integration is started, there is a string printed to the logs, this includes the model and serial number (these are case sensitive). it looks something like this: 

[image]

Cheers!

pjhum, 2024-11-02:
Apologies, I'm fairly new to HA. which log, exactly? I've searched <settings> <system> <logs> and there is no entry adding an inverter.

Peter 

From: Joshua Leaper 
Sent: Saturday, November 2, 2024 6:24 AM
To: CharlesGillanders/homeassistant-alphaESS 
Cc: pjhum ; Mention 
Subject: Re: [CharlesGillanders/homeassistant-alphaESS] "Alpha ESS: Set Battery Charge" not setting (Issue #138)

Sorry @pjhum should of been more specific. when the integration is started, there is a string printed to the logs, this includes the model and serial number (these are case sensitive). it looks something like this:

image.png (view on web)

Cheers!

pjhum, 2024-11-02:
OK- removed the integration and re-added. The start up message was 
Alpha ESS Energy Statistics : AL7002022040171
Storion-SMILE-B3 (AlphaESS) 
that looks like the same string ?

My version still reports 0.5.5 BTW
Peter

Poshy163, 2024-11-02:
To view logs in Home Assistant, you can navigate to Settings > System > Logs then click on view all logs.

If you reboot the integration, click ctrl + f (page finder thing) and look for "Model". It should bring up that line I posted earlier

Poshy163, 2024-11-02:
Also, to force upgrade to 5.6.0, 

Under the integrations page on HACS, you can click the RELOAD DATA button under the SETTINGS tab (should be like the top right of the page) and then an update will appear.

Maybe do this and verify you can't set the discharge/charge via the buttons before i blacklist that model inverter

braindeadmalc, 2024-11-02:
> Hey there, do you have the full name (the model of it) so I can add it to the blacklist? Should show in your logs as Model: xxx Serial: xxxx

Mine's "New Inverter: Serial: AL4001020050581, Model: Storion-SMILE-B3"

pjhum, 2024-11-02:
Thanks, I now have 0.5.6 showing.The new controls have appeared.They do not appear to work, though. I pressed the "reset charge discharge button" and my AlphaEss app reports no change in settings.I'm presuming it should be instant, or nearly so?What I don't understand if the hardware is incompatible is that it was happily letting me set the overnight charge% until the last Alpha update/outage. PeterSent from Android device

Poshy163, 2024-11-02:
Yeah, if you click on any of the time periods, you then should be able to open your app and see the time settings set. Cheers for that all!

Poshy163, 2024-11-03:
Cheers for the logs, ill add the inveter to the blacklist and push a minor update soon, the current/discharge charging time (1 & 2), and the charging range within the diagnostics section of the stuff will stay, just your ability to set anything (within the Configuration area) will be removed

braindeadmalc, 2024-11-03:
> Cheers for the logs, ill add the inverter to the blacklist and push a minor update soon, the current/discharge charging time (1 & 2), and the charging range within the diagnostics section of the stuff will stay, just your ability to set anything (within the Configuration area) will be removed

Ah, I have a support ticket open with Alpha. If they fix the API issue will I be able to check it works?

Poshy163, 2024-11-03:
ill keep it as a draft PR until you hear back, ive also talking with one of the guys behind the API to see if its intended or not.

Do keep us updated 👍🏻

pjhum, 2024-11-03:
Thanks; I really need Alpha to fix this otherwise I am going to have to change energy tariffs! 

Resetting the charge/discharge data manually in the app makes the unit very little useto me.

I'm clinging to the belief that they broke this accidentally in the last upgrade rather than deprecating the older inverters without notice...though neither is a recommendation of the company.

Many thanks for all the hard work on this integration- it had served me very well.

Peter

Poshy163, 2024-11-03:
All good @pjhum, just curious to see if this works, if set the charge/discharge data manually in the app (for some time) does it show up in the integration in the config area (you may need to restart the integration, or wait a few mins)

pjhum, 2024-11-03:
Yes, I can see the charge/discharge settings and they updated pretty quickly (2-3 min max) in the 'diagnostic' card of the integration to match a change made in the app. I can't tell what _batHighCap or _batUseCap refer to in the 'config' secrtion, but they don't change. I had assumed they were to do with battery protection in the background. 

Peter 

From: Joshua Leaper 
Sent: Sunday, November 3, 2024 11:37 AM
To: CharlesGillanders/homeassistant-alphaESS 
Cc: pjhum ; Mention 
Subject: Re: [CharlesGillanders/homeassistant-alphaESS] "Alpha ESS: Set Battery Charge" not setting (Issue #138)

All good @pjhum, just curious to see if this works, if set the charge/discharge data manually in the app (for some time) does it show up in the integration in the config area (you may need to restart the integration, or wait a few mins)

Poshy163, 2024-11-03:
Here is from the API docs: 
batUseCap | Discharging Cutoff SOC [%]
batHighCap | Charging Stops at SOC [%]

Its basically the "bounds" for how much the batteries connected can charge and discharge, this is sent to the API when one of the buttons are pressed, obv if the set discharge and charge calls arnt working, i dont think it would be doing much

pjhum, 2024-11-03:
OK- so it should set the same values that are reported in the diagnostic card. That is the part that isn't happening...

Peter 

From: Joshua Leaper 
Sent: Sunday, November 3, 2024 12:07 PM
To: CharlesGillanders/homeassistant-alphaESS 
Cc: pjhum ; Mention 
Subject: Re: [CharlesGillanders/homeassistant-alphaESS] "Alpha ESS: Set Battery Charge" not setting (Issue #138)

Here is from the API docs:
batUseCap | Discharging Cutoff SOC [%]
batHighCap | Charging Stops at SOC [%]

Its basically the "bounds" for how much the batteries connected can charge and discharge, this is sent to the API when one of the buttons are pressed, obv if the set discharge and charge calls arnt working, i dont think it would be doing much

Poshy163, 2024-11-03:
Well, yeah. which likely means that either you haven't pressed the charge/discharge (and waited up to a min for the next fetching of the API), orrr the setcharge and setdischarge isn't working, which seems to be the case here, i have asked for clarification by am yet to hear back

Poshy163, 2024-11-03:
The batUseCap *should* also be the minimum amount the battery can discharge (its set as the Discharging cut off SOC in the settings) 

while the batHighCap is only shown as the Charging stops at SOC setting (aka is only used for force charging the battery)

pjhum, 2024-11-03:
Yes, that is clear. The different names in the integration were confusing me, but a moment's thought tells me they need to be unique... 

I previously used the  Alpha ESS Set Battery Charge successfully in my automations to set summer night charging based on Solcast forecasts.

I was trying to set up & test an automation to stop my battery emptying into the car on Intelligent Octopus using the Alpha ESS Set Battery Discharge call when I realized neither was working... TBH the 'new' controls are less important to me than the existing ones resuming.

I have reached out to Alpha UK- without holding out much hope.

thanks again for your work and explanations.
Peter 

From: Joshua Leaper 
Sent: Sunday, November 3, 2024 12:33 PM
To: CharlesGillanders/homeassistant-alphaESS 
Cc: pjhum ; Mention 
Subject: Re: [CharlesGillanders/homeassistant-alphaESS] "Alpha ESS: Set Battery Charge" not setting (Issue #138)

The batUseCap should also be the minimum amount the battery can discharge (its set as the Discharging cut off SOC in the settings)

while the batHighCap is only shown as the Charging stops at SOC setting (aka is only used for force charging the battery)

ppootle, 2024-11-03:
I have a very similar issue: Setting the charge and discharge times worked perfectly via the integration until about 2 weeks ago, but no longer seems to work. Only reading the diagnostics now seems to work via the integration. Manually setting the charge and discharge time via the app still works - and then shows correctly in the diagnostics. Have removed and re-added the integration but can't get it to work to set the timings anymore.

Poshy163, 2024-11-04:
@ppootle do you also have a Storion-SMILE-B3?

Poshy163, 2024-11-04:
@pjhum you might have some luck with alpha support, 
[image]
translates to: "This charge and discharge setting is applicable to all models. If it does not work, you need to submit a customer complaint and contact our after-sales service."

ppootle, 2024-11-04:
No. It's a Smile5INV.

pjhum, 2024-11-04:
Thanks @Poshy163, I'll wait and see when they get back to me.

Was that a Google translate job or can you read Chinese? If so, I'm impressed...

renamedthewild, 2024-11-04:
I cant get the discharge setting to work with t10HV or Smile5

Poshy163, 2024-11-04:
Haha, I used google translate, I'm not that good.

Also, did some more digging. Seems to be a widespread issue. See the video below. Setting the discharge time seems to disable it? I'll do some digging as to if it was something on my end. Or the API

https://github.com/user-attachments/assets/7d74ccd6-8f63-4157-a331-7907c3e6bfa4

Poshy163, 2024-11-04:
Ah well, found the issue. if anyone needs an fix now, and knows how to get to the source files of home assistant,  

On line 93 of the coordinator.py, there is a 0 in the updateDisChargeConfigInfo call, change to a 1, save. then reboot and it will work. otherwise and update will go out as soon as Charles approves my PR

renamedthewild, 2024-11-04:
Can this fix be done at a home asssit level ?

Poshy163, 2024-11-04:
you can use the visual studio code add-on, but i would not recommend it if you are not sure, rather wait for the PR to be merged

renamedthewild, 2024-11-04:
Got it

renamedthewild, 2024-11-04:
> Ah well, found the issue. if anyone needs an fix now, and knows how to get to the source files of home assistant,
>
> On line 93 of the coordinator.py, there is a 0 in the updateDisChargeConfigInfo [...]

I made the change, however it did not seem to resolve the issue

pjhum, 2024-11-04:
Cheers- sheer coincidence that I only started to try and control discharge after the sun had gone in for the year in Lancashire, so didn't see the link! 

I'll wait for the merge- I'm not a PYTHON fan...

Thanks again for all your effort on this.

Peter 

From: Joshua Leaper 
Sent: Monday, November 4, 2024 9:40 AM
To: CharlesGillanders/homeassistant-alphaESS 
Cc: pjhum ; Mention 
Subject: Re: [CharlesGillanders/homeassistant-alphaESS] "Alpha ESS: Set Battery Charge" not setting (Issue #138)

you can use the visual studio code add-on, but i would not recommend it if you are not sure, rather wait for the PR to be merged

Poshy163, 2024-11-04:
> I made the change, however it did not seem to resolve the issue
assuming you have restarted home assistant after making the change, and waiting 10 mins (since the last press as there is a cooldown implemented by alphaess)

Does anything show in the app? the fix worked for me, and i can show a video verifying if you'd like?

i would just recommend waiting for the patch and then giving it a go

Poshy163, 2024-11-04:
Interesting. i would wait for the .1 release, and update. if this still occurs you can make another issue. it could be case of you have changed the wrong variable.

DavidqStokes, 2024-11-04:
I am a little unclear here about the scope of this fix.
Does this fix the new force charge/discharge feature only, or does it also fix the wider issue of the failure to set charge/discharge parameters via automations?

Poshy163, 2024-11-04:
> Does this fix the new force charge/discharge feature only

It only fixes the features offered by this integration, so if there is any automations tied to it (that uses this). Not sure what external providers like Amber use tho

DavidqStokes, 2024-11-04:
I'll check what happens when my automations run (trigger the actions to make the set charge / discharge) this evening

Poshy163, 2024-11-04:
The charge should work, the discharge wont until the latest PR is merged

pjhum, 2024-11-04:
I’ve had a response from Alpha UK- they suggest trying using a different URL for the API-https://apiservice-eu.alphaess.com/v1/Open/XXXXX

I don’t know which URL the integration uses. If it isn’t this I need to find some tools to interrogate the API manually, I gues…

Peter

ppootle, 2024-11-04:
I got this response from UK support: 

"Could you please update the URL and API address to below and try again?
European Center URL: https://apicloud-eu.alphaess.com/
European Center API address: https://apiservice-eu.alphaess.com/v1/Open/XXXXX
"

Not sure what i need to do with it though. I get user "user does not exist" if i try the URL.

pjhum, 2024-11-04:
You need an API test tool- Postman or whatever.

I've downloaded SoapUI which is free but I'm struggling to frame a request (any request) to test it.  The API documentation is just that, not a tutorial... plus I've never used the software before. 

I'm on some other training tomorrow but will have a dig later in the week if I get time- though for it to help in HA the source code for the integration would have to be updated anyway. Thinking about it, that might be the easiest test path.

Poshy163, 2024-11-04:
The update has been pushed, so you can force the  fetch the new update by going onto the integration (on the HACS page), click on the three dots (⋮) on the integration’s card and then update information

Poshy163, 2024-11-04:
> "Could you please update the URL and API address to below and try again? European Center URL: https://apicloud-eu.alphaess.com/ European Center API address: https://apiservice-eu.alphaess.com/v1/Open/XXXXX "

Interesting, we dont keep the baseURL (aka the URL to the API) anywhere in this repo, its kept in the openAPI repo [here](https://github.com/CharlesGillanders/alphaess-openAPI)

Changing that as the end user is not easy, and i would recommend if you want to test it, you use something like postman (there is a example collection in that openAPI repo) 

Maybe try the .1 release that just got put out and see if that fixes it. otherwise might need to re-work where the base URL is deprived from.

[40 more replies]