CharlesGillanders/homeassistant-alphaESS #139: AlphaESS contains no entities

darkpainy, 2024-10-23:
Hi, 

I have an issue where I re-added my AlphaESS integration, setup is successfull but there are no entities found.
In the log I see this error: (unexpected keyword arugment "model_id"

2024-10-18 11:57:21.755 ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform alphaess
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 616, in _async_add_entity
    device = dev_reg.async_get(self.hass).async_get_or_create(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'model_id'
2024-10-18 11:57:21.759 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up alphaess platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 368, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 616, in _async_add_entity
    device = dev_reg.async_get(self.hass).async_get_or_create(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'model_id'
[image]

Poshy163, 2024-10-23:
This might be due to the flakyness with the API, are you able to use postman (the data collection is here: https://github.com/CharlesGillanders/alphaess-openAPI/blob/main/AlphaESS%20Open%20API.postman_collection.json)

and get me some of the results? typically that happens due to not being able to call `getEssList`

dukevyner, 2024-11-09:
I'm having the exact same issue, please excuse the ignorance, how exactly do we "use poastman"?

dukevyner, 2024-11-21:
> This might be due to the flakyness with the API, are you able to use postman (the data collection is here: [...]

i am still having this issue, i have tried reseting the api dev key, deleting and re adding it to th api site.
deleting and reinstalling HA, HACS, and the aplhaESS intigration.

i managed to figure out how to use the postman json.

my results for the getEssList are as follows:

`{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": [
        {
            "sysSn": "ALD001122121035",
            "popv": 10570.00,
            "minv": "SMILE-G3-S5-INV",
            "poinv": 5.00,
            "cobat": 10.08,
            "mbat": "SMILE-G3-BAT-10.1P",
            "surplusCobat": 9.58,
            "usCapacity": 95.00,
            "emsStatus": "Normal"
        }
    ],
    "extra": null
}`

any insight as to how to resolve this issue would be greatly appreicated

Poshy163, 2024-11-22:
what error are you receiving @dukevyner when you try to start up the integration? are you on 5.6.1 and is there any other errors within the log that is shown before this error is made? 

There may be an issue with getting the serial number, which is the model_id and then it "links" all the services from a single SN together, maybe one of the calls are failing for some reason in the integration

dukevyner, 2024-11-22:
@Poshy163 thanks for the reply
Currently running 0.5.6.1
Intigration  completes showing "[image]" after entering the appid and secret key

One entry is recorded with the appid as the name

However [image] or entities are added

(excuse the mobile away from home atm) 

There are 4 [image] logged
All recorded below in the order of which they occurred 

```Logger: homeassistant.components.button
Source: helpers/entity_platform.py:580
integration: Button (documentation, issues)
First occurred: November 21, 2024 at 8:04:34 PM (28 occurrences)
Last logged: 12:33:21 PM

Error adding entity None for domain button with platform alphaess
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 580, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 750, in _async_add_entity
    device = dev_reg.async_get(self.hass).async_get_or_create(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'model_id'
```

```
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:580
integration: Sensor (documentation, issues)
First occurred: November 21, 2024 at 8:04:34 PM (136 occurrences)
Last logged: 12:33:21 PM

Error adding entity None for domain sensor with platform alphaess
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 580, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 750, in _async_add_entity
    device = dev_reg.async_get(self.hass).async_get_or_create(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'model_id'
```

```
Logger: homeassistant.components.number
Source: helpers/entity_platform.py:580
integration: Number (documentation, issues)
First occurred: November 21, 2024 at 8:04:34 PM (8 occurrences)
Last logged: 12:33:22 PM

Error adding entity None for domain number with platform alphaess
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 580, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 750, in _async_add_entity
    device = dev_reg.async_get(self.hass).async_get_or_create(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: DeviceRegistry.async_get_or_create() got an unexpected keyword argument 'model_id'
```
```
Logger: homeassistant
Source: runner.py:147
First occurred: November 21, 2024 at 8:04:20 PM (6 occurrences)
Last logged: 12:33:22 PM

Error doing job: Unclosed client session
Error doing job: Unclosed connector
```

Poshy163, 2024-11-22:
Interesting, 
when you are on PC, you should see some logs like this when you start up the integration. then a final data, are you able to take a screenshot of all of this (i have included what mine looks likes). it should be within the logs after you restart/enter in your info. then no entities appear.

[image]

dukevyner, 2024-11-22:
@Poshy163  interstingly mine is not showing the trying to call, parts but it is fetching some of the data including the serial and model
 
[image]
[image]

the main errors repet a few times after this with no change

dukevyner, 2024-11-22:
@Poshy163 sorry, clearly i had not enabled logging, figruing this out as we go. here is what you actually wanted
[image]