CharlesGillanders/homeassistant-alphaESS #140: cant reinstall/configure this integration

simonr23, 2024-10-25:
my integration was stuck on 'not loaded' - a strange state. it wouldnt allow deletion from within the standard settings menu, so i deleted it via HACS.

i redownloaded the newest version and tried to re-add it as an integration, but am now stuck gettign an error 500 response the moment i select the integration from the '+' option.

i've tried numerous restarts, deletions, versions. no joy.

Poshy163, 2024-10-25:
This might be due to the flakyness with the API, are you able to use postman (the data collection is here: https://github.com/CharlesGillanders/alphaess-openAPI/blob/main/AlphaESS%20Open%20API.postman_collection.json)

and get me some of the results? typically that happens due to not being able to call `getEssList`

simonr23, 2024-10-27:
> This might be due to the flakyness with the API, are you able to use postman (the data collection is here: [...]

i have no idea how to do that, sorry. i tried googling, but no joy.

simonr23, 2024-10-30:
update - i managed to get it setup via using my iphone app, and not the webpage. maybe just a coincidence with alpha also coming online? but either way - sorted.

closed 2024-12-18 by CharlesGillanders