CharlesGillanders/homeassistant-alphaESS #141: Missing EV charger power and state

farmer1969, 2024-10-25:
Hello Charles,

I'm missing the actual EV charger power along with charger state like:

[image]

Is that something you could implement?

Thanks,
Frank

CharlesGillanders, 2024-10-25:
The Alpha OpenAPI at https://open.alphaess.com/developmentManagement/apiList does have getEvChargerStatusBySn  which returns the charger state, there's also getEvChargerCurrentsBySn  - does that return the power metric that you are looking for?

I don't have either of those currently used in my python library, I don't have a charger to test with but if you would be open to volunteering to do some testing I'm sure that I could add both of them relatively easily.

farmer1969, 2024-10-25:
Certainly I could help here as I have an AlphaESS Charger type 'SMILE-EVCT11'. And I have experience in using the ioBroker adapter provided by Gaspode69 at https://github.com/Gaspode69/ioBroker.alpha-ess/tree/main . Since version 1.3.0 this is supporting the wallbox based on OpenAPI.

farmer1969, 2024-10-26:
I will see whether I can work on it on Sunday. Today, no chance...

Rasenkardinal, 2024-11-07:
This would be really great! I am happy to help with testing if necessary.

farmer1969, 2024-11-07:
Hi @CharlesGillanders, @Poshy163, I almost forgot that I provided the details (response from OpenAPI) some time back already to Gaspode69 in Storion4You forum. Based on the information available [here ](https://www.storion4you.de/thread/1004-test-aufruf-an-die-alpha-ess-wallbox-besitzer-und-iobroker-nutzer/?postID=8268#post8268) he implemented the EV charger in the [ioBroker AlphaESS adapter](https://github.com/Gaspode69/ioBroker.alpha-ess). Would this information be sufficient? I guess so.
If required I would translate that from German to English but Google Tranlator should do was well...

Poshy163, 2024-11-12:
Cheers for that @farmer1969 so the only data you are looking at using is the  "currentsetting"?

farmer1969, 2024-11-13:
Not really, I think I should extend the information...

Please refer to the script in [#13](https://www.storion4you.de/thread/1004-test-aufruf-an-die-alpha-ess-wallbox-besitzer-und-iobroker-nutzer/?postID=8261#post8261). This covers most use cases I believe, meaning all relevant ev-charger APIs.

As there are (refer to https://open.alphaess.com/developmentManagement/apiList):

- getEvChargerConfigList ==> Obtain the SN of the charging pile according to the SN, and set the model
- getEvChargerCurrentsBySn ==> Obtain the current setting of charging pile household according to SN
- setEvChargerCurrentsBySn ==> Set charging pile household current setting according to SN
- getEvChargerStatusBySn ==> Obtain charging pile status according to SN+charging pile SN
- remoteControlEvCharger ==> According to SN+ charging pile SN remote control charging pile to start charging/stop charging
- getChargeConfigInfo ==> According SN to get charging setting information
- updateChargeConfigInfo ==> According SN to Set charging information，Setting frequency 24 hours, set once a day

All that shall be provided through the AlphaESS integration for use through HA.

farmer1969, 2024-11-29:
Hi @CharlesGillanders, @Poshy163, is the above anything you can work with?
Thanks, Frank

Kavino112, 2024-12-01:
Guten Tag, ich würde mich gerne als Tester zur Verfügung stellen.

Ich besitze die SMILE-G3-EVCT11

Poshy163, 2024-12-04:
Hey there, sorry for the late response, 

Yeah im able to implement this, thing is, its going to make the startup time for the integration very long, as alpha has enforced a limit of 10 API calls per minute, and currently we are at 8 on startup, adding these 4 will bring it to 12, which means those 12 will need to be throttled at 10 per min, so it will take ~1:12 to fully start up the integration which home assistant does not like at all (throws slow start up warnings and hangs HA until setup is complete)

I am currently working on a PR to add this stuff, and i also have a ticket open with alpha to see if we can get the rate limit eased a bit. but not sure how successful that will be as the API team that develop it/maintain it seem to be in limbo at the moment

farmer1969, 2024-12-04:
Isn't there any option to proceed for the startup as is (with only 8 API calls) and set a time-based trigger to do the rest 1-2 min later...?