CharlesGillanders/homeassistant-alphaESS #145: Please add epvtotal from the summary data as sensor

paulbusse, 2024-11-12:
Hello,

I use the epvtotal to notify me that it is (about) time to request for a new certificate (Belgian thing).
I found that  using daily generation does not work if the Alphaess api is not available. Using the total picks up when the api is accessable again. E.g.

Day 1: daily 10 kWh; total 5000kWh
Day 2: [API down] (but a production of 10kWh)
Day 3: [API down] (but a production of 10kWh)
Day 1: daily 10 kWh; total 5040kWh

Using the day production I would miss the 20kWh from days 2 and 3.
Using the total production I still can see that the production of my solar panels is 40kWh.

Thanks

Poshy163, 2024-11-12:
@paulbusse the PR is open, there is still a few other sensors that need to be added (in relation to EV charging) and then a new version will be released. if you know how to manually add code to the HACS integration in feel free to through the custom_components folder, otherwise it should be released soon*

paulbusse, 2024-11-12:
Thanks and sorry for missing the PR.

Poshy163, 2024-11-12:
Haha no stress, you didnt miss any PR, i made it shortly after you made this issue 😄

closed 2024-12-19 by CharlesGillanders