CharlesGillanders/homeassistant-alphaESS #149: Request Vitapower Control

IncorporateApps, 2024-12-01:
Do you think there will be support for the Vitapower Powerstation?
Specifically to see battery state and also set the AC charge and discharge values via the AC connection?

Then you can connect this to a smart meter and fill it with PV power from the AC grid.

Poshy163, 2024-12-01:
I guess it is possible for us to integrate it assuming all the parameters are sent via the API, are you able to test it at all?

IncorporateApps, 2024-12-01:
Unfortunately not at the moment.

eekhoorn, 2025-01-01:
I'm interested as well in this feature: I want to able to start/stop charging and decharging the Vitapower with a variable Wattage via HomeAssistant. 
In the current api specs of  AlphaEss I cannot see the possibility the set the Wattage, only the api calls make it possible to start/stop charging/decharging.

Furthermore:
@IncorporateApps : do you have a smart socket that can limit the power? I only have smart sockets like Shelly Plus Plug S that can switch on/off and monitoring the power, but limiting the power is not possible.

IncorporateApps, 2025-01-15:
> I'm interested as well in this feature: I want to able to start/stop charging and decharging the Vitapower with a variable Wattage via HomeAssistant. In the current api specs of AlphaEss I cannot see the possibility to set the Wattage, only the api calls make it possible to start/stop charging/decharging.
> 
> Furthermore: @IncorporateApps : do you have a smart socket that can limit the power? I only have smart sockets like the Shelly Plus Plug S that can switch on/off and monitoring the power, but limiting the power is not possible.

Do you have the Vitapower? Can you test it and maybe help @Poshy163 with the integration?

eekhoorn, 2025-01-15:
Yes, I will test the HA integration when I have received the dutch version of the Vitapower (called Evapower).
I expect the delivery before the end of January.

IncorporateApps, 2025-01-15:
> Yes, I will test the HA integration when I have received the dutch version of the Vitapower (called Evapower). I expect the delivery before the end of January.

Great, keep us updated. As long as there is HA integration, it should work better.

eekhoorn, 2025-01-28:
Yesterday I received the Vitapower VT1000 or EvaPower as branded in NL. 
Good news: the HA Integration is working: 44 entities are created!

Some sensors are not relevant for this unit. E.g. sensor.XXX_instantaneous_ppv3 and sensor.XXX_instantaneous_ppv4 because only 2 ppv are supported.
Also the button.XXX_YY_minute_charge are not working in my case, but I have to test it further. 
If I can help to test / develop it further, please let me know.

IncorporateApps, 2025-01-28:
Can you change the AC charging power setting of the Vitapower via HA- API script?  And also see the battery state in %? Thank you

IncorporateApps, 2025-01-28:
> Battery state: yes: sensor.XXX_instantaneous_battery_soc is reporting correctly
> 
> Change the AC charging power setting: I have to test the actions "Alpha ESS: Set Battery Discharge" and "Alpha ESS: Set Battery Charge" further.

Ok, thank you. Can you at least change the AC charging power in the App?

eekhoorn, 2025-01-28:
Yes, in the app it is possible to change: 1) Charge: 20 - 2000W 2) Discharge: 0 - 800W or to 2000W after approval from Alphaess

efmuurman, 2025-01-30:
I bought the EvaPower in the netherlands. ( same as the vitapower ) 
reading data works. But for example the function to charge 15 minutes doesn't work.

IncorporateApps, 2025-01-30:
> function to charge 15 minutes doesn't work

Please explain this function in detail, maybe attach some screenshots, thank you!

Poshy163, 2025-01-31:
I've been thinking about it, i really don't think a product like this would be suitable to add into this current integration, which focuses on home batteries + solar. Rather we move it into its own integration 

Both with (i assume) Charles and i not having this a vitapower, its very hard to test if its even feasible to add to the existing codebase. coupled with the previous issues around their API (more specifically the API limits around the amount of calls that can be made per minute being low) 

I would be more than happy to make a fork of this, and move it to a separate vitapower integration, which focuses on it and its abilities. i would just need someone's appid and app secret (or better yet a unit 🙈) and i could see how it goes.

IncorporateApps, 2025-01-31:
How many calls do they allow per minute, is 5-6 ok?

Poshy163, 2025-01-31:
The current limit appears to be 8, the integration uses 6. with the additional stuff required from the vitapower it seems like we will need 9? which we could spread to ~1 a few secs, like what currently happens if you have more than a single inverter linked to your OpenAPI account. 

Still doesn't help that there is 0 way for myself to test the functionality (aka what calls are supported and what isnt) which would be needed to not releases a broken buggy mess

efmuurman, 2025-01-31:
VitaPower / Evapower are both homebatteries from AlphaEss and partly the API works.

IncorporateApps, 2025-01-31:
> the integration uses 6.

What do you need 6 for? What if I don't need what you need? If we make the API work so that we can change the AC charge speed and the DC discharge speed and this every 10 seconds (so 6) without other calls, would that be enough? 

Basically what I need is to tell the Vitapower to charge via AC when the panels are producing more than the house uses, and to discharge when the PV are not producing enough, but the house demands it.

Or do you need something else ontop of that, because if not 6 is fine with me?

Are those two things easy to implement?

Poshy163, 2025-02-01:
The biggest issue, API restrictions aside,  is the lack of myself having one, and not being able to test it, if you want, you can send me a message on discord (poshy163) and send my your app id and app secret and i can have a look at the feasibility around it and get a prototype working

> What do you need 6 for?

The OpenAPI has many calls for different things, there is not a single API call that can retrieve all the required info (see the image) 

[image]

DavidqStokes, 2025-02-01:
What is different about the Vitapower?

The reason I ask is that my Smile B3 (AC connected) is set to charge overnight, and to discharge during the day.

However, it will still charge during the day is we have excess solar (it monitors PV production and grid import/export). Does the Vitapower lack this monitoring?

eekhoorn, 2025-02-01:
> The biggest issue, API restrictions aside, is the lack of myself having one, and not being able to test it, if you want, you can send me a message on discord (poshy163) and send my your app id and [...]

I'm willing to share app id & app secret for a certain period of time. I'm not a big fan of discord.  I prefer Telegram.

Poshy163, 2025-02-01:
Sure, telegram works, @poshy6945

eekhoorn, 2025-02-01:
See Telegram messageMvg Eekhoorn

IncorporateApps, 2025-02-01:
Thank you guys, if we can make this work with about 6 calls a minute, that would be amazing
"Basically what I need is to tell the Vitapower to charge via AC when the panels are producing more than the house uses, and to discharge when the PV are not producing enough, but the house demands it."

eekhoorn, 2025-02-01:
Yes, or in terms of power sensors: "When my power.grid sensor >= 0: go  discharging and when my power.grid sensor < 0: go charging"

IncorporateApps, 2025-02-01:
More importantly, we have to set the exact power in Watts for both, AC and DC discharge. Just simply tell it to discharge-charge with preset values is not enough.

eekhoorn, 2025-02-01:
Charge/discharge values should be dynamically adjusted: 
X: if power.grid sensor >= 0 then go discharging with [power.grid sensor value]  Watt, E.g power.grid sensor = 450W

X+ 10 seconds:  if power.grid sensor >= 0 then go extra discharging with [current power.grid sensor value]  Watt, E.g  current power.grid sensor = 30W => total discharge value should be 450 + 30 = 480W etc.

mauricebish, 2025-02-09:
I have also got the EvaPower. Your integration works. In a networktrace I saw a channel being active on TCP port 6667 (IRC) from the app to the main unit. The communication was encrypted.

Poshy163, 2025-02-20:
Ive made the repo: [https://github.com/Poshy163/homeassistant-VitaPower](https://github.com/Poshy163/homeassistant-VitaPower)

i still need some API credentials, so would be good if anyone could send me theirs so i can do some testing (telegram: @poshy6945 or discord: poshy163)

eekhoorn, 2025-02-20:
Check your telegram account

Poshy163, 2025-02-20:
Alright, turns out might not need its own integration. seems like all of the calls are the same, but with additional data. should be a easy merge

closed 2025-02-21 by CharlesGillanders

IncorporateApps, 2025-02-21:
Thank you! Can someone please provide their scripts for doing the things listed here, read AC charge and discharge and set accordingly, with the correct calls and confirm all works correctly and can be executed every 10 seconds without problems by the API?

CharlesGillanders, 2025-02-21:
> Thank you! Can someone please provide their scripts for doing the things listed here, read AC charge and discharge and set accordingly, with the correct calls and confirm all works correctly and can be executed every 10 seconds without problems by the API?

There's a limit on setting frequency in the API, that's an alpha restriction not something any of the home assistant developers have set.  Documentation says it's limited to once per day, I don't recall if that's accurate or not right now.

Poshy163, 2025-02-22:
I think they changed it to every 10-15 mins? Whatever it's currently set to in the integration is the last known restriction i got from alpha. 

I'm currently working on hopefully a work around to it, but it's prolly going to be a bit before it releases (alongside support for the wallbox and hopefully charge and discharge with these 2 things) which requires an update to the openAPI (which is this PR here https://github.com/CharlesGillanders/alphaess-openAPI/pull/15) 

Will keep you guys updated, but at the moment it's looking promising

Poshy163, 2025-02-28:
@IncorporateApps @eekhoorn @efmuurman:

Received this response from alpha:

Q: Are you able to control the charge and discharge of the VitaPower AC charging power and the DC discharge speed via the API?
A: VitaPower's API has not yet been opened in OpenAPI.

Not much can be done until they "open"/release it, i can keep an eye out from time to time, but knowing alpha, its going to take a while unfortunately

mauricebish, 2025-02-28:
I have asked the owner of EvaPower if he can speed up things. He has forwarded the request to AlphaESS. He will let me know if there is news. I am in direct contact with EvaPower to improve and solve issues with the EnergyLife app.

IncorporateApps, 2025-03-02:
> Q: Are you able to control the charge and discharge of the VitaPower AC charging power and the DC discharge speed via the API? A: VitaPower's API has not yet been opened in OpenAPI.
>
> Not much can be [...]

Thank you, but I thought that you can still use it via a "closed" API? Didne maurice send you the logs and I thought you said it is doable and already working?

PR #166 (2025-02-21): Improve VitaPower support and Add additional sensors 
Bumps version to 0.5.8 (will be released when the wallbox is supported in the next PR)

Adds sensors: 
- Electric Vehicle Power One
- Electric Vehicle Power Two
- Electric Vehicle Power Three
- Electric Vehicle Power Four
- pev
- pmeterDc
- PrealL1
- PrealL2
- PrealL3
- Inverter nominal Power
- Pv nominal Power
- Battery model

Removes Charge/Discharge settings from all VitaPower (VT1000) devices

resolves #149