CharlesGillanders/homeassistant-alphaESS #152: Setup failed for custom integration 'alphaess': Unable to import component: No module named 'alphaess'

Eweol, 2024-12-15:
Hi guyes,

I get since some days the following Error

`Setup failed for custom integration 'alphaess': Unable to import component: No module named 'alphaess'`

this is the full log:

`2024-12-15 13:48:20.684 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'alphaess': Unable to import component: No module named 'alphaess'
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
component = await integration.async_get_component()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/loader.py", line 1034, in async_get_component
self._component_future.result()
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
File "/usr/src/homeassistant/homeassistant/loader.py", line 1014, in async_get_component
comp = await self.hass.async_add_import_executor_job(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
self._get_component, True
^^^^^^^^^^^^^^^^^^^^^^^^^
)
^
File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 58, in run
result = self.fn(*self.args, **self.kwargs)
File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
ComponentProtocol, importlib.import_module(self.pkg_path)
~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
return func(*args, **kwargs)
File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
return _bootstrap._gcd_import(name[level:], package, level)
~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
File "<frozen importlib._bootstrap_external>", line 1022, in exec_module
File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
File "/config/custom_components/alphaess/__init__.py", line 8, in <module>
from alphaess import alphaess
ModuleNotFoundError: No module named 'alphaess'`

Homeassistant Version: 2024.12.3
AlphaESS Version: 0.5.6.1
Infrastructure: k8s Container
Install: HACS

Can someone help?

Poshy163, 2024-12-15:
Try uninstalling and re-installing. alphaess is the external package to manage communication with the API, for some reason it might have been deleted?

Eweol, 2024-12-16:
Hi @Poshy163,

already tried that several times

CharlesGillanders, 2024-12-16:
When home assistant installs any custom modules it pulls the required libraries from PyPi.org - if there's any DNS issue or firewall blocking traffic between your HA install and PyPi the alphaess library won't be installed.

Is there any indication in your logs or error messages of any issue with downloading that library?  Or any network or dns issues in general?

Eweol, 2024-12-17:
Hi,

No there is nothing like that in logs, as well as it already worked fine till mid of november I think

CharlesGillanders, 2024-12-18:
You may need to delete the integration from HACS and then manually make sure that it is completely removed from the custom_components folder in your Home Assistant install before trying the re-install.  It's possible that some fragment of the original install is still hanging around and getting Home Assistant confused about what it needs to do to deploy the component.

Eweol, 2024-12-18:
Today I fully uninstalled, removed the repo and all dependent files, without luck.
I get still the same error message.

Eweol, 2024-12-26:
Hi @CharlesGillanders ,

thx for your help.
After some investigations, I found out the problem is not the integration itself, but the pip command which should install the integration. This seems to fail. I installed it manual within my container and now its working again.

Is that something the integration has control about or is this a failure by homeassistant?

CharlesGillanders, 2024-12-26:
> Is that something the integration has control about or is this a failure by homeassistant?

Glad you got sorted, the integration relies on home assistant for all of the underlying python infrastructure.  Integration developers don't even call pip directly we just tell home assistant what libraries are needed and it's supposed to do the rest.

Eweol, 2024-12-26:
Sure, so I will close that issue and create one for Homeassistant.

Thx again

closed 2024-12-26 by Eweol