CharlesGillanders/homeassistant-alphaESS #155: verification code by postman

rk-code, 2024-12-19:
Hello, I tried to get the verification code for registering my inverter on open.alphaess by using postman as descirbed in the documentation. To register directly at open.alphaess did not work, see #153.
Unfortunatly I get after sending getVerificationCode with my serial number and check code filled in corrcetly in the variables as a return
{
    "code": 6007,
    "msg": "Sign check error"
}
Do you have any idea whats going wrong ?

Tumblehawk79, 2024-12-20:
getting the same 6007 response from the postman send api

CharlesGillanders, 2024-12-20:
Looks like we need to further update the documentation.  As well as your serial number and check code you need to add your app id and app secret from the alphaESS developer portal into the variables for the collection.  Should be placeholders there already for those variables just as there was for serial and check code.

Poshy163, 2024-12-21:
Will do that now 👍🏻, once the updated read.md is merged, ill make a new hacs release and that should address all the current issues.

Tumblehawk79, 2024-12-22:
I think we may have a deeper issue with the Alpha Developer portal.
when signing up for Developer portal it no longer assigns/displays a AppID or Appsecret (the field seems to get a internal 500 error), so unable to try the verification process unless I'm missing something basic, I'm new to this.

when i view developer portal:
[image]

when i run the alpha developer portal verification get internal 500 error (known issue)
when i run postman api I get the 6007 sign check error (which i know is due to the missing ID and secret)  

so i don't think any real issue with your stuff, just seems their portal is broken.

Poshy163, 2024-12-22:
Yeah, wow. i mean you could try contacting alphaess support, and see what they say, but im not really sure who its responsible for the API/developer stuff anymore

Tumblehawk79, 2024-12-22:
I logged a service ticket under my device, will see where it goes.

Tumblehawk79, 2024-12-23:
# shellyn.chua
2024-12-23 01:13:25
Dear Customer, 
 
Thank you for bringing this issue to our attention. We have escalated your case to our API team for further investigation. Once the issue is resolved, we will contact you with an update.
 
We appreciate your patience and understanding. Should you have any further questions in the meantime, please feel free to reach out.
 
Kind regards,
Team Alpha ESS

Tumblehawk79, 2024-12-23:
seems to be fixed for me, my appID and appsecret appeared and was able to add my battery (the open.alphaess.com website seems to be working fine even did the verification code correctly, no 500 errors)

i did get about 8 verification messages all at once so suspect they had a backlog once they cleared the issue.  

HA install and configuration worked, can see data from battery in HA.

Poshy163, 2024-12-23:
Nice, i suspect this will fix most of the issues people have been happening with onboarding to the OpenAPI, glad to see they are fixing it pretty promptly after it being reported

PR #156 (2024-12-22): Update README.md
Fixes #155
closed 2024-12-23 by Poshy163