CharlesGillanders/homeassistant-alphaESS #158: Integration not working after HA Core update

Rasenkardinal, 2025-01-10:
Hi there,

I updated to the new HA-Core V 2025.X.X and get a setup error for the integration. So it doesn't work anymore.
I tried several times to reinitialize and to reinstall the integration but the result is always as described.

If I return to the earlier version 2024.X.X  via backup everything is working fine again. 

Ist this a known issue or is it just me?

DavidqStokes, 2025-01-10:
I am on 2025.1.1 (I skipped 2025.1.0)

The integration is working fine for me, both in reading data and setting charge/ discharge times for my Smile B3 battery.

I have seen no problem with the integration initialising after restarts.

Poshy163, 2025-01-10:
Anything in the logs? been working fine all through the betas

Rasenkardinal, 2025-01-10:
I was not up to date with the integration. My fault! 

Everything is fine.

Sorry an thanks for the quick answers.

closed 2025-01-10 by Rasenkardinal