CharlesGillanders/homeassistant-alphaESS #160: values are not getting updated

TiSpace, 2025-01-14:
Hi,
using AlphaEss 0.5.7. HA Core 2025.1.1, HA OS 12.4

All of my AlphaESS entities (as far as I can see) are found and updated once HA is new started. However then I'm not getting any data update. Is there anything I need to enable to run the data acquisition?

Poshy163, 2025-01-15:
Anything within the logs?

TiSpace, 2025-01-15:
I have
- enabled debug protocol within the AlphaEss integration
 => just a warning after HA restart ("...integration has not been testes by HA...") but not related to my problem

- checked protocol (Setting-> Protocol -> HA Core)
   => no Alpha related errors or warning 

So (from my point of view) nothing in the logs

TiSpace, 2025-01-15:
ok, after spending days in finding the issue, it seems to work (again). 
What I did:
- installed the very latest HA updates (Core+OS) -> at this time it was not working
- tried to deinstall the integration, only thing I was able to delete were the integration service and added my battery again
- 
So far it seems to work. I have to say that I tried all single steps in the past, which was not rewarded by a positive result. Still some mysteries left.