CharlesGillanders/homeassistant-alphaESS #161: Configuration step using HACS

ianmc-HB, 2025-01-16:
I have installed the integration using the steps listed in "Installation using HACS"
Steps 1-5 completed successfully
At step 6 the configuration fails with:

_Error
Config flow could not be loaded: 500 Internal Server Error Server got itself in trouble__

The Homeasistant.log shows
2025-01-16 10:17:10.566 ERROR (SyncWorker_0) [homeassistant.util.package] Unable to install package
......  rows deleted  form log
......  last 2 rows in the log

  File "/snap/home-assistant-snap/637/lib/python3.12/site-packages/homeassistant/requirements.py", line 307, in _async_process_requirements
    raise RequirementsNotFound(name, list(failures))
homeassistant.requirements.RequirementsNotFound: Requirements for alphaess not found: ['alphaessopenapi==0.0.11']. 

the only reference to "alphaessopenapi"  is in the manifest.json file
"requirements": [
    "alphaessopenapi==0.0.11"

I have not found any other ref to "alphaessopenapi", so i do not know what I have to do to get the configuration steps to complete.

system details:  on old hp6730b laptop for trial and testing
Ubuntu 24.04.1 LTS
Home assistant\ Core 2024.10.4  Frontend 20241002.4
HACS 2.03
AlphaESS Energy Storage System v0.5.7

Poshy163, 2025-01-16:
Heya @ianmc-HB , 

This *could* be related to the older version of HA core you are running. just to rule out this as a possibility. can you uninstall the integration, and upgrade to the newest version and then try to reinstall? The newest current version is: 
> Core 2025.1.2
Supervisor 2024.12.3
Operating System 14.1

As for what "alphaessopenapi" is, its the python library used to interface between the home assistant add-on, and the cloud platforms, you can see here: https://pypi.org/project/alphaessopenapi/

Home assistant is meant to download these on boot, would there be any reason your HA instance cant reach this website to download it from the internet?

ianmc-HB, 2025-01-17:
Thank you for the advice, I shall upgrade the core to 2015.1.2 etc., and will post any issues/progress here.
I am able to access the URL to the alphaessopenapi

[image]