CharlesGillanders/homeassistant-alphaESS #162: Unable to configure AppId etc

Kevin2460-kn, 2025-01-19:
Since last HA update my Alpha data has gone. I have tried to reinstall but can't get it to accept my Appid and scret. Have tried solution offer and get verification code 6007 unsucessful.
Any suggestions would be appreciate. Running on HA green
cheers Kevin

Kevin2460-kn, 2025-01-19:
unsuccesful should read "sign check error

Poshy163, 2025-01-19:
Heya @Kevin2460-kn,

what version of the integration are you currently on? and when you uninstalled and reinstalled it, did you remove the integration, uninstall via HACS, restart your HA instance, and the install the integration via HACS and then restart again, then adding your info to the integration via the add device section 

typically a sign check error occurs when the appid/secret is wrong or blank, so the HA instance could of lost the details somehow and without a full removal of the integration, its kept something wrong

Kevin2460-kn, 2025-01-19:
Sounds like I may have only partially uninstaller. I reinstalled the latest
version on HA Green effectively I thought starting from scratch. Reset my
TP Link modem and then re downloaded Alpha and  reinstalled but seemed
there must be something left, I will try your suggestion and let you know
what happens. I am new at HA so you prompt reply is muck appreciated
Cheers Kevin

Kevin2460-kn, 2025-02-05:
Have uninstalled and reinstalled several times but can't get past unexpected error when I enter AppId etc. have tried postmaster but still no go.

CharlesGillanders, 2025-02-05:
If you are getting an error in postman then it's probably not the integration that's broken. It's something wrong with alpha.  Can you post the details of your postman call and the error that postman is giving you (minus the sensitive data)