CharlesGillanders/homeassistant-alphaESS #167: Entity sensor.albxxxxxxxxxxxx_solar_to_battery from integration alphaess has state class total_increasing, but its state is negative.

dmutsaers, 2025-02-22:
Just wanted to report the following warning I see in my log file:

Entity sensor.albxxxxxxxxxxxx_solar_to_battery from integration alphaess has state class total_increasing, but its state is negative.

CharlesGillanders, 2025-02-22:
Thanks, it's a known issue; the Alpha ESS API occasionally returns values that are basically wrong, that's outside of our control.  We perhaps could do a better job of handling when this happens but for now this is just a warning that home assistant is showing and has no impact.

xDAVIDx93, 2025-02-26:
I wonder if removing the inverter from your open.alphaess.com dashboard and re-adding it would fix it?

Poshy163, 2025-02-26:
I doubt, it's just rouge/incorrect data being gathered from the API (on alphas side) 

We could theoretically restrict the data shown to only be a positive number, but in the past have decided against a client side restriction, a few of us agreed (with relation to another issue) it's better to just display the true and most accurate data received from the API, rather than artificially changing it