CharlesGillanders/homeassistant-alphaESS #169: requirement missing when updating to 0.6.0.0

mipo1972, 2025-03-14:
It seems the new release does not contain the alphaessopenapi version 0.0.13.
see the following error after updating via hacs

redownload of the previous versions and all works again.

the error in the logs:
2025-03-14 20:01:47.129 ERROR (SyncWorker_4) [homeassistant.util.package] Installed version for alphaessopenapi resolved to None
2025-03-14 20:01:48.203 ERROR (SyncWorker_3) [homeassistant.util.package] Installed version for alphaessopenapi resolved to None
2025-03-14 20:01:48.597 ERROR (SyncWorker_3) [homeassistant.util.package] Unable to install package alphaessopenapi==0.0.13: × Failed to read `alphaessopenapi==0.0.13`
  ├─▶ Failed to read metadata from installed package `alphaessopenapi==0.0.13`
  ├─▶ Failed to parse METADATA file:
  │   `/usr/local/lib/python3.13/site-packages/alphaessopenapi-0.0.13.dist-info/METADATA`
  ╰─▶ Metadata field Name not found
2025-03-14 20:01:48.817 ERROR (SyncWorker_3) [homeassistant.util.package] Unable to install package alphaessopenapi==0.0.13: × Failed to read `alphaessopenapi==0.0.13`
  ├─▶ Failed to read metadata from installed package `alphaessopenapi==0.0.13`
  ├─▶ Failed to parse METADATA file:
  │   `/usr/local/lib/python3.13/site-packages/alphaessopenapi-0.0.13.dist-info/METADATA`
  ╰─▶ Metadata field Name not found
2025-03-14 20:01:49.058 ERROR (SyncWorker_3) [homeassistant.util.package] Unable to install package alphaessopenapi==0.0.13: × Failed to read `alphaessopenapi==0.0.13`
  ├─▶ Failed to read metadata from installed package `alphaessopenapi==0.0.13`
  ├─▶ Failed to parse METADATA file:
  │   `/usr/local/lib/python3.13/site-packages/alphaessopenapi-0.0.13.dist-info/METADATA`
  ╰─▶ Metadata field Name not found
2025-03-14 20:01:49.064 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'alphaess': Requirements for alphaess not found: ['alphaessopenapi==0.0.13'].

Poshy163, 2025-03-14:
Interesting, it seems like the package has installed, but you are unable to read it. 

Can you try re-installing 0.6.0? Seems like that something might of gone wrong with home assistant

mipo1972, 2025-03-14:
Hi,
 
retried several times, no success. with HA restarts in between. no luck.

rebooted the full system, installed the previous version
updated again, now it works.

Michiel

closed 2025-03-14 by mipo1972