CharlesGillanders/homeassistant-alphaESS #172: Enable history / statistics for relevant entities such as total income and battery capacity

x4N70pHyLL, 2025-03-25:
Hello,

can we have the statistics / history enabled for the relevant entities?

That would be in my opinion at least

- Total Income
- Self sufficiency
- Self consumption
- Current Capacity (of the Battery)
- Maximum Battery capacity

As it makes sense to monitor them long-term, in some cases long term only.

For kWh entities this is already enabled (I assume this is a HA default depending on the state_class)

regards

x4N70pHyLL, 2025-03-25:
Yes exactly!

Poshy163, 2025-03-25:
Yep, linked a PR. will test it and confirm thee LTS are saved, i know there is some special rules around what needs to be defined for LTS to work: 

[https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics](https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics)

ill merge the PR when its confirmed it works