CharlesGillanders/homeassistant-alphaESS #174: Set the entity batusecap does not work anymore

realloki, 2025-03-28:
Hello Charles.

The entity batusecap can't be set on a SMILE-S6-HV-INV.

I use this setting to set the backup power reserve depending on solcast prediction. If it's lower than 9 kWh a day and the actual SoC is higher than 30 % I set batusecap to 30 %. In other cases I set it to 11 %.

When I set this value in HA and log in to cloud.alphaess.com the value is not set. The life entities like _instantaneous_battery_soc does show actual values und they are updated.

What do you need for further inspection? How can I assist to find the cause?

Thank you in advance
realloki

Poshy163, 2025-03-28:
Does the other one (sethighcap) work at all? Does it reflect in the alphaESS app?

realloki, 2025-03-29:
Hi Poshy163.I can’t see this setting in the app. Maybe its not exposed for me. Within HA I can see it‘s set to 90%Is this behind an switch to expose this setting?Setting battsecap has worked about 3 or 2 weeks ago. I tested this setting a value and check in the app if it‘s set. And this happend.ServusSteffan

Poshy163, 2025-03-30:
I said this in #175, but i feel like Alpha may have changed a few things around recently, and introduced/reduced certain features working. 

Just to confirm. on your device page, next to your serial number. it shows SMILE-S6-HV-INV? ill make write a new email to alpha about it

realloki, 2025-03-30:
Hi Poshy163.

Yes. The model is a:
Systemmodell : SMILE-S6-HV-INV
When I got this delivered this model didn't show up on the AlphaESS website. Maybe I got one of the first models. I got this from Energieversum. A german company who managed the whole installation. They do 10000 installations per year. The contract was for an Smile Hi 5 inverter.

I would like to see a setting where I can set the times when to charge the battery so that the high peaks in summer could be cut off at high noon. Also combined with predictions from solcast. I thought of charging early in the morning, stopping until around clock 10 or 11 so that the peak for the grid is lowered.

I just see a setting where I can define times when to charge from grid. Newly with dynamic energy prices.

realloki, 2025-03-30:
On the web interface I can see 'Charging Cutoff SoC %'. But this ist just for charging from grid. This setting is only accessible when 'Enable Grid Charging Battery' is enabled.
Below that there is the option for 'Enable Battery Discharge Time Control' with the time settings. Also inaccessible if time control is not enabled.
Below that there is the option 'Discharging Cutoff SOC [%]'
I only use the 'Discharging Cutoff SOC [%]' functionality.