CharlesGillanders/homeassistant-alphaESS #175: Charge and Discharge Sensors Stuck at 0.0 kWh for VT1000 Home Battery

frederikhardeman, 2025-03-29:
**Problem**
The Charge (sensor.algxxx_discharge) and Discharge sensors (sensor.algxxx_charge) in Home Assistant remain at 0.0 kWh for a VT1000 home battery. However, sensors for real-time usage of this battery (such as instantaneous_battery_i_o) are still functioning properly. These sensors were working on March 25 but stopped updating on March 26.

The battery % (sensor.algxxx_instantaneous_battery_soc) is still working, and charging and discharging of the battery itself are also functioning. No changes were made to Home Assistant in the meantime. Reloading the integration or restarting HA didn't help.

**HA Version**
2025.03.4

Does anyone know the cause and a possible solution for this issue?

Poshy163, 2025-03-30:
Interesting, i feel like AlphaESS has updated what is supported from the API for each device. I have someones VT1000 for testing, and it appears to be also on 0, but it has always been. 

Ill send alpha an email about it and see whats going on, seems like they have changed a few things around recently

JohnRabbit1, 2025-04-14:
Yesterday it starts working again.= (13-4)

frederikhardeman, 2025-04-14:
Interesting, because my values are still 0,0.

JohnRabbit1, 2025-04-15:
Hi,

 

I have made a screendump this morning. As you can see the Battery was loaded in the night.

1 point of interest: Last Sunday i had the impression that the battery was restarted somehow. I had to reconnect it to my network (is a bit complex).

I did not pay very much intention to this. But seeing you and other persons with still 0.0 values, maybe it is a good idee to check for updates and restart?

eversf, 2025-04-15:
It's working for me now. Had to restart the VT1000.

[image]

Poshy163, 2025-04-15:
Maybe a new firmware update was installed? and all it took was rebooting to push it or something? strange....

frederikhardeman, 2025-04-15:
I rebooted the battery and reloaded the integration, still no values, but let's see what happens tomorrow. Will update my findings.

frederikhardeman, 2025-04-16:
I can now also report that after a restart of my battery yesterday, I got values back again! Of course after restart they will be 0, but the counting of charge or discharge immediately starts after that.

4nuts, 2025-04-25:
Same problem here , also no charge or discharge info in energy dachboard; after a reboot of the Eva Battery and reload integration it started again , after half hour it shows the charge and discharge again. sensor.chardge and sensor.discharge

Poshy163, 2025-04-30:
@4nuts is there anything in the logs relating to it? would be curious to see if there was maybe another model of the VitaPower VT1000 that has different parameters and it doesnt work

Poshy163, 2025-04-30:
Ah yeah, did think it was somewhat a firmware issue on the devices. 

ill close this issue, and if a new firmware update breaks it, feel free to re-open an issue

4nuts, 2025-05-02:
> Fyi: again the values didn't work for me, out of a sudden around 28 April. I restarted the battery again today, and the values are back again.

Fyi: Here every thing is running fine. Charge and Discharge is correct . I Use the " Own Use P1 meter" profile.

Poshy163, 2025-05-25:
Heya, yeah would be cool if you could share those stats as well. but also if you don't mind. open a ticket with alpha, as i strongly believe its *something* on their end that is causing this

nderooij, 2025-05-26:
Thanks @Poshy163 !!
I have asked EvaPower if this is something they can help with as well, as I just noticed that within the "Energy Life" app on the "Week -> Battery" section, the vallues also stopped showing after a specific date.
Let me see if they can fix it or have a faster way in with AlphaESS.

Meanwhile, I noticed that on the sticker of my EvaPower it says VT1000, but in the HA Integration it shows I've got "Battery Model" VM1000. I'm thinking VM is the EvaMate and VT is the EvaPower, but I wouldn't want to waste your time, so rather mention it now.

Please find the output of my Postman below:

`
GetEssList
{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": [
        {
            "sysSn": "ALG000000000000",
            "popv": 2.00,
            "minv": "VT1000",
            "poinv": 2.00,
            "cobat": 4.10,
            "mbat": "VM1000",
            "surplusCobat": 3.90,
            "usCapacity": 95.00,
            "emsStatus": "Protection"
        }
    ],
    "extra": null
}

----------------------------------------------------------------------------------------------------------------------------------------------------------------
GetVerificationCode
{
    "code": 6003,
    "msg": "You have bound this SN!",
    "expMsg": null,
    "data": null,
    "extra": null
}

----------------------------------------------------------------------------------------------------------------------------------------------------------------
BindSN
{
    "timestamp": "2025-05-26T15:15:55.172+00:00",
    "status": 405,
    "error": "Method Not Allowed",
    "path": "/api/bindSn"
}

----------------------------------------------------------------------------------------------------------------------------------------------------------------
GetSumData
{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": {
        "epvtoday": 0.00,
        "epvtotal": 171.00,
        "eload": 0.00,
        "eoutput": 0.00,
        "einput": 0.00,
        "echarge": 0.00,
        "edischarge": 0.00,
        "todayIncome": 0.0,
        "totalIncome": 0.00,
        "eselfConsumption": 0.19,
        "eselfSufficiency": -1.14628,
        "treeNum": 29.2773,
        "carbonNum": 147.0600,
        "moneyType": null
[... 2837 lines cut ...]
----------------------------------------------------------------------------------------------------------------------------------------------------------------
UpdateDisChargeConfigInfo
{
    "code": 200,
    "msg": "Success",
    "expMsg": null,
    "data": {
        "ctrDis": 0,
        "timeDisf1": "00:00",
        "timeDise1": "00:00",
        "timeDisf2": "00:00",
        "timeDise2": "00:00",
        "batUseCap": 15.00
    },
    "extra": null
}

----------------------------------------------------------------------------------------------------------------------------------------------------------------

----------------------------------------------------------------------------------------------------------------------------------------------------------------

----------------------------------------------------------------------------------------------------------------------------------------------------------------

----------------------------------------------------------------------------------------------------------------------------------------------------------------
`

Poshy163, 2025-06-14:
Heya @nderooij thanks for letting me know! 

glad to see they are taking kindly to some feedback for once 😄, if it happens again, feel free to open a new issue/re-open this one

closed 2025-06-14 by Poshy163

basdp, 2025-08-08:
Here with the EvaPower/ZinVolt system still no valies on the last update version. It has worked before. 

How do you restart the system so I can check if that changes anything?