CharlesGillanders/homeassistant-alphaESS #183: Two Inverters Installations; only four entities available in HA:Integration Entries

daudus, 2025-05-05:
Hello.

I have two independent inverters installed in my house. Both are from AlphaESS: "SMILE-T10-HV-INV" and "Smile5-INV".
After the installation of the AlphaESS I can see two services/inverters, but only 4 services:  twice "batHighCap" and "batUseCap".

My HAOS is installed in Proxmox 8.3.3 as a virtual machine on a Raspberry Pi 5 with 16 GB RAM.

Home Assistant version info:
Core 2025.4.4
Supervisor 2025.04.1
Operating System 15.2
Frontend 20250411.0

The attachment contains responses from OpenAPI tests; one file per inverter. I've run only the "get" methods without any issues.

_Note: I had to modify the [here](https://www.postman.com/poshy163/alphaess/collection/tsy43t1/alphaess-open-api?action=share&creator=11219653) the post-response script in the "getEssList" method, because it does not count with more than one inverter (see "pm.collectionVariables.set("SysSn", jsonData.data[**0**].sysSn);")_

[APITest.Smile5-INV.2025-05-01.txt](https://github.com/user-attachments/files/20032930/APITest.Smile5-INV.2025-05-01.txt)
[APITest.SMILE-T10-HV-INV.2025-05-01.txt](https://github.com/user-attachments/files/20032929/APITest.SMILE-T10-HV-INV.2025-05-01.txt) 

Many thanks for helping me

David

Poshy163, 2025-05-05:
Heya @daudus, 

Do you have an image of what it looks like on your home assistant instance? 

does the 2 services of the batHighCap appear on only the Smile5-INV? or the SMILE-T10-HV-INV? 

if im understanding this, you have 2 separate alphaESS AppID amd AppSecret? one for each pair (1x Smile5-INV and 1x SMILE-T10-HV-INV)?

daudus, 2025-05-05:
Hi [Poshy163](https://github.com/Poshy163). 

1. Cloud.Alphaess.Com.png contains a view of AlphaESS Cloud. I have one AppID and one AppSecret for both inverters.
[image]

2. The following images demonstrate the view directly in HA: 

[image]

[image]

[image]

daudus, 2025-05-05:
Undertand. But problem is not related to naming of these four entities/sensors - problem is missing other entities/sensors. I can see only these four now!

Poshy163, 2025-05-05:
Ahh, was late last night when i was looking at that and i didn't notice that 😅 

@daudus can you possibly try and re-install and re-add your devices to it? alpha cloud might have been having some issues at the time and they might have been fixed by now.

if that doesn't work, are you able to send me your app id and app secret over telegram, my username is @poshy6945 and i can check out if its an issue somehow with how the devices are being added in