CharlesGillanders/homeassistant-alphaESS #184: API Data Refresh Rate

TimHillman17, 2025-05-08:
Hi,

Great to get this connectivity in to HA, thank you!

What is the refresh rate of the data through the API?  It seems a lot slower than the native Alpha App.
Is there a parameter to adjust it somehow?

Thanks.

Poshy163, 2025-05-09:
Heya @TimHillman17,

Indeed your right about it being slower, it currently polls the api once every 60s.

if you would like to change it, within the `custom_compoents/alphaess/const.py` file you will see on line 13 a `SCAN_INTERVAL = timedelta(minutes=1)`, you are able to modify it to be shorter if you would like (eg `SCAN_INTERVAL = timedelta(seconds=30)` but i wouldn't recommend going below 20 or so seconds as you may be rate limited by alpha

TimHillman17, 2025-05-13:
> Poshy163
>  left a comment
> (CharlesGillanders/homeassistant-alphaESS#184)

That’s brilliant - thank you.

Tim.

closed 2025-05-13 by Poshy163