CharlesGillanders/homeassistant-alphaESS #186: The workaround to get Verification Code doesn't work

OscarrrJ92, 2025-05-12:
Hi there,

I tried to follow your guide to connect to the AlphaCloud since I have an AlphaESS battery through this link: https://github.com/CharlesGillanders/homeassistant-alphaESS. I tried the workaround for getting a Verification Code, but I noticed in the https://www.postman.com/poshy163/alphaess/request/uns7ntg/getverificationcode?tab=scripts code that this is not the getverificationcode script, but looks like a copy of the Discharge script. 

Can you check this for me please and correct it if neccessary, so I could get with the correct script a verification code. This would help me a lot! :)

Hope you can set it right. Thanks in advance!

Best regards,

Oscar

CharlesGillanders, 2025-05-13:
@Poshy163 This looks to be something broken in your existing postman collection.  Would you be able to take a look?

Poshy163, 2025-05-13:
Heya @OscarrrJ92. should be fixed now, give it a go

OscarrrJ92, 2025-05-13:
Heya @Poshy163,

Thanks for the fast response! Unfortunately I still get "200" as a return. I guess something in the "Scripts" of /getVerificationCode still doesn't go well. I added a printscreen of the forked collection. Can you please check this?

Thank you very much!

[image]

Poshy163, 2025-05-13:
Ah apologies, updated that now

Poshy163, 2025-05-13:
Strange, as it seems to be working here with no issues, maybe its worth going to alphaess support to ask? 

[image]

[image]

OscarrrJ92, 2025-05-13:
Hi @Poshy163 

I followed the exact steps, forked the collection, filled the variables, but unfortunately still get this message:

[image]

I'll contact AlphaESS. Thank you very much for your time and patience! :)