CharlesGillanders/homeassistant-alphaESS #190: Reversing sensor.instantaneous_grid_i_o_total

TimHillman17, 2025-06-06:
Hi - apologies if this is totally the wrong forum....please feel free to shut me down! :-)

How can I invert the value of the sensor.instantaneous_grid_i_o_total?

I have garage which I want to show the level of export, but the number is negative.

Thanks.

TimHillman17, 2025-06-06:
Thanks - that's helpful.

Context:  on the Alpha iOS app, there is a feed-in number, which I "assume" is sensor.instantaneous_grid_i_o_total.

I want my gauge to show my export, not my net-import (which is currently negative as I'm generating more than my load).

Poshy163, 2025-06-06:
Could do something like this with a Left Riemann sum Helper, see this: [https://www.home-assistant.io/integrations/integration/](https://www.home-assistant.io/integrations/integration/)

i have mine set up something like this: 

[image]

[image]